Moddable-OpenSource/moddable #815: stack-overflow xsMemory.c:640 in fxMarkInstance

jessysaurusrex, 2022-01-24:
### Version: 
[eecb298fb1d9019854591515664e369ad9cf5b53](https://github.com/Moddable-OpenSource/moddable/tree/eecb298fb1d9019854591515664e369ad9cf5b53)

### poc.js
```js
function main() {
async function v0(v1,v2,v3,...v4) {
    const v5 = await v2;
    const v7 = new Promise(v0);
    const v8 = v7.finally(v5);
}
const v9 = v0();
}
main();
```

### Invocation `xst poc.js`
```
STDERR: 
/Users/amulet/moddable/xs/sources/xsScript.c:203:16: runtime error: left shift of 1327365517 by 1 places cannot be represented in type 'txSize' (aka 'int')
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/amulet/moddable/xs/sources/xsScript.c:203:16 in 
AddressSanitizer:DEADLYSIGNAL
=================================================================
==90879==ERROR: AddressSanitizer: stack-overflow on address 0x00016d50fda0 (pc 0x000102427870 bp 0x000102420198 sp 0x00016d50fdc0 T0)
    #0 0x102427870 in fxMarkInstance xsMemory.c:640

SUMMARY: AddressSanitizer: stack-overflow xsMemory.c:640 in fxMarkInstance
==90879==ABORTING

Status: 
pid 90879 SIGABRT (signal 6)
```

phoddie, 2022-01-26:
Garbage collector overflows the native stack while recursively marking promises. This will be guarded against with calls to `fxCheckCStack`.

jessysaurusrex, 2022-02-02:
Verified fixed in `fb7b2c2`

closed 2022-02-02 by jessysaurusrex