Moddable-OpenSource/moddable #818: Async generator not closed when yielding rejected promise

mhofman, 2022-01-25:
**Build environment:** Linux
**Target device:** XS Test Engine 10.5.0

**Description**
An async generator yielding a rejected promise should trigger a `finally` clause when the iterator gets closed.

**Steps to Reproduce**
```js
globalThis.print ||= (...args) => console.log(...args);

(async () => {
  async function* iterator() {
    try {
      yield Promise.reject("reject");
    } finally {
      print("A");
    }
  }

  try {
    for await (const x of iterator());
  } catch (e) {
    print("B", e);
  }
})();
```

**Expected behavior**
```
A
B reject
```

**Current behavior**
```
B reject
```
**Other information**
I noticed that XS deviates from other engine implementations when working on related [test cases](https://docs.google.com/presentation/d/1W9EJoWOvi6jU1A2bwyixzOceoP_ASAeaOYER3Zy9P00/edit) highlighting a [normative PR regarding closing an async wrapped iterator](https://github.com/tc39/ecma262/pull/2600)

phoddie, 2022-01-25:
Thanks for the report. We'll take a look.

phoddie, 2022-02-01:
The behavior has been updated to match the referenced PR. The test script provided now passes. Please verify. Thank you!

closed 2022-02-09 by phoddie