Moddable-OpenSource/moddable #819: Makefile paths are corrupt

cmidgley, 2022-01-25:
**Build environment:** Linux
**Target device:** ESP32

**Description**
Similar to the bug reported earlier (#775) where parts of the filename are removed from the path, I have a more extreme version happening now. The prior issue fails to generate a valid `tsconfig.json` file, whereas this problem has a valid `tsconfig.json` but generates a corrupt `makefile`.

I am using multiple manifest files, across multiple scoped modules.  The module name in this case is `@cmidgley/jest-runner` who in turn consumes `@cmidgley/jest-recorder`.   The makefile fails when attempting to compile as the contents of the makefile have lost several characters of the filename:

```
$(MODULES_DIR)/@cmidgley/jest-recorder.xsb: $(MODULES_DIR)/ecorder/src/main.js
    @echo "# xsc @cmidgley/jest-recorder.xsb"
    $(XSC) $(MODULES_DIR)/ecorder/src/main.js -d -c -e -o $(@D) -r jest-recorder
$(MODULES_DIR)/@cmidgley/TestTracking.xsb: $(MODULES_DIR)/ecorder/src/TestTracking.js
    @echo "# xsc @cmidgley/TestTracking.xsb"
    $(XSC) $(MODULES_DIR)/ecorder/src/TestTracking.js -d -c -e -o $(@D) -r TestTracking
```

You can see here where `jest-recorder` gets shorted out to `ecorder`.  This same file also corrupts `jest-runner` into `unner`.

I've tried a wide range of filenames, and discovered very little!

```
test-tracking -> tracking
testracking -> tracking
testrun-tracking -> run-tracking
xyzzy -> xyzzy (this is the only one so far that worked!)
jest-tracking -> tracking
jest-recorder -> ecorder
```

It's possible I made some errors during those tests, as I have to hand edit around 20 files and it takes a while to get it all working again each time.

If I include this exact top level manifest file from another manifest, it generates perfectly fine:

```
$(MODULES_DIR)/@cmidgley/jest-recorder.xsb: $(MODULES_DIR)/jest-recorder/src/main.js
	@echo "# xsc @cmidgley/jest-recorder.xsb"
	$(XSC) $(MODULES_DIR)/jest-recorder/src/main.js -d -c -e -o $(@D) -r jest-recorder
$(MODULES_DIR)/@cmidgley/TestTracking.xsb: $(MODULES_DIR)/jest-recorder/src/TestTracking.js
	@echo "# xsc @cmidgley/TestTracking.xsb"
	$(XSC) $(MODULES_DIR)/jest-recorder/src/TestTracking.js -d -c -e -o $(@D) -r TestTracking
```

**Manifest files**

`@cmidgley/jest-runner/manifest.json`:
```json
{
  "include": [
    "node_modules/@cmidgley/di/manifest.json",
    "$(MODDABLE)/examples/manifest_typings.json",
    "$(MODDABLE)/modules/base/modules/manifest.json",
    "node_modules/@cmidgley/jest-recorder/manifest.json"
  ],
  "build": {
    "BUILD": "$(MODDABLE)/build",
    "MODULES": "$(MODDABLE)/modules",
    "COMMODETTO": "$(MODULES)/commodetto",
    "PIU": "$(MODULES)/piu"
  },
  "modules": {
    "*": [
      "$(MODULES)/files/resource/*",
      "$(MODULES)/base/instrumentation/modInstrumentation"
    ],
    "@cmidgley/jest-runner/TestRunner": ["src/TestRunner"],
    "@cmidgley/jest-runner/TestFramework": ["src/TestFramework"]
  },
  "typescript": {
    "compiler": "$(MODDABLE)/moddable-tsc",
    "tsconfig": {
      "compilerOptions": {
          "types": [
            "$(MODDABLE)/xs/includes/xs",
            "src/TestFramework"
          ]
      }
    }
  },
  "defines": {
    "XS_MODS": 1
  },
  "preload": ["Resource"],
  "strip": "*",
  "platforms": {
    "esp32": {
      "include": "$(BUILD)/devices/esp32/manifest.json",
      "modules": {
        "*": "$(MODULES)/network/socket/lwip/modLwipSafe"
      }
    }
  },
  "creation": {
    "static": 131072,
    "chunk": {
      "initial": 1536,
      "incremental": 512
    },
    "heap": {
      "initial": 512,
      "incremental": 64
    },
    "stack": 256,
    "keys": {
      "available": 32,
      "name": 53,
      "symbol": 3
    },
    "parser": {
      "buffer": 1024,
      "table": 17
    },
    "main": "main"
  }
}
```

@cmidgley/jest-recorder/manifest.json:
```json
{
  "build": {
    "BUILD": "$(MODDABLE)/build",
    "MODULES": "$(MODDABLE)/modules",
    "COMMODETTO": "$(MODULES)/commodetto",
    "PIU": "$(MODULES)/piu"
  },
  "include": [
    "$(MODDABLE)/examples/manifest_typings.json",
    "$(MODDABLE)/modules/files/file/manifest.json"
  ],
  "modules": {
    "@cmidgley/jest-recorder": ["src/main"],
    "@cmidgley/TestTracking": ["src/TestTracking"]
  },
  "creation": {
    "static": 131072,
    "chunk": {
      "initial": 1536,
      "incremental": 512
    },
    "heap": {
      "initial": 512,
      "incremental": 64
    },
    "stack": 256,
    "keys": {
      "available": 32,
      "name": 53,
      "symbol": 3
    },
    "parser": {
      "buffer": 1024,
      "table": 17
    },
    "main": "main"
  },
  "preload": ["Resource"],
  "strip": "*",
  "platforms": {
    "esp32": {
      "include": "$(BUILD)/devices/esp32/manifest.json",
      "modules": {
        "*": "$(MODULES)/network/socket/lwip/modLwipSafe"
      }
    }
  },
  "typescript": {
    "compiler": "$(MODDABLE)/moddable-tsc",
    "tsconfig": {
      "compilerOptions": {
        "types": ["$(MODDABLE)/xs/includes/xs"]
      }
    }
  }
}
```

**Steps to Reproduce**
Currently unable to reproduce in a small system.  Any hints or ideas on what to try let me know.  I'm dead in the water unable to build right now.

cmidgley, 2022-01-25:
Got it!  It appears there is some type of length limited string compare going on, where it fails when two modules have similar starting portions of their names (perhaps scope being required, did not test that as I always used scope).

Steps:
1) Clone [this repo](https://github.com/cmidgley/moddable-makefile-common-root)
2) cd into `common-one`
3) mcconfig -m -d -p esp32

Fingers crossed you also get this error:

```
# xsc timer.xsb
# xsc Resource.xsb
# xsc mc/config.xsb
# tsc tsconfig.json
# xsc @common-two/common.xsb
#  file not found: /root/Projects/moddable/build/tmp/esp32/debug/common-one/modules/two/common.js!
make: *** [/root/Projects/moddable/build/tmp/esp32/debug/common-one/makefile:643: /root/Projects/moddable/build/tmp/esp32/debug/common-one/modules/@common-two/common.xsb] Error 22
```

And when looking in the generated `makefile`:

```
$(MODULES_DIR)/@common-two/common.xsb: $(MODULES_DIR)/two/common.js
	@echo "# xsc @common-two/common.xsb"
	$(XSC) $(MODULES_DIR)/two/common.js -d -c -e -o $(@D) -r common
$(MODULES_DIR)/@common-one/main.xsb: $(MODULES_DIR)/one/main.js
	@echo "# xsc @common-one/main.xsb"
	$(XSC) $(MODULES_DIR)/one/main.js -d -c -e -o $(@D) -r main
$(MODULES_DIR)/@common-one/common.xsb: $(MODULES_DIR)/one/common.js
	@echo "# xsc @common-one/common.xsb"
	$(XSC) $(MODULES_DIR)/one/common.js -d -c -e -o $(@D) -r common
%two/common.js %one/main.js %one/common.js : %/tsconfig.json
	@echo "# tsc tsconfig.json"
	tsc -p $(MODULES_DIR)/tsconfig.json
```

Notice that it references `one` and `two` in the directory paths, rather than `common-one` and `common-two`. 

In my tests, I found if I change to a different root of the module name, in this case rename `common-one` to perhaps `unusual-one` it will then generate a makefile and compile correctly.  The number of characters that need to be the same appear to be around 6 or 7.

phoddie, 2022-01-31:
Just a quick note to confirm that I'm able to reproduce the problem. No clue yet about what is going on.

phoddie, 2022-01-31:
I have a proof-of-concept fix  that seems to work. To test, I made a couple changes to your example project.

First, the `modules` section of common-one/manifest.json changes to move main out of @common-one:

```json
  "modules": {
    "@common-one/*": ["./common"],
    "*": "./main"
  }
```

Second, main.js changes to validate that the two modules import correctly:

```ts
import {common} from "@common-one/common";
import {altcommon} from "@common-two/common";

trace(common, "\n");
trace(altcommon, "\n");
```

The change is to mcmanifest.js, lines [425-430](https://github.com/Moddable-OpenSource/moddable/blob/e55fbbf8c9a450acb052fc1630171f04a11c25c2/tools/mcmanifest.js#L425-L430):

```js
				const stop = Math.min(first.lastIndexOf("/"), last.lastIndexOf("/"));
				common = 0;
				while ((common < stop) && (first.charAt(common) === last.charAt(common))) 
					common++;
```

Would you give that a try and let me know how it goes? If it solves your problem, we can sort out a proper fix which I *think* involves stopping at the first directory mismatch.

cmidgley, 2022-01-31:
Thanks for the time and effort, Peter.  I tried it with the sample repo and it works.  Then I tried it with my real code (migrated my code back to the similar names for scoped modules), and it failed.

I spent a few hours trying to reproduce this in the smaller repo, and finally figured out it has to do with path depth.  Perhaps you are eluding to that with your comment about "stopping at the first directory mismatch"? I updated the repo to show the failure by adding another level of directories in case you want to see it / test with a different use case.

I ran the test without moving `main`, as that is strongly my preferred solution (which works fine with the small repo).   Each of my modules has it's own main, under the scoped module name.  In most cases these provide the imports for the module but in some cases (and in particular in this one that is failing) the `main` is actually a different named file which starts up automated unit testing.  Much of this deals with the fact that the modules are shared across Node and Moddable, resulting in naming and scoping limitations.    Hopefully I will be able to leave `main` as it is, pointing to the scoped module, as that works well outside of this issue and solves several issues for me.

cmidgley, 2022-01-31:
Doh!  And of course I already destroyed that environment...  It's rebuilt again and the repo is updated (this time with actual source files!).  Same as before, build from `common-one`.  And yes, I do use a `creation` on all my real (top level) `manifest.json` files and specify `main` in there.  A small `creation` is defined inside `common-one/manifest.json` file to define `main`.

phoddie, 2022-01-31:
I hope you didn't delete the environment again, because all the `.ts` files are empty (0 bytes).

cmidgley, 2022-01-31:
Oh come on ... I give you files, and that wasn't enough?  Now you want contents??

Multiple stupid human errors, sorry.  Fixed (including removing and cloning the repo to test that it works for me).

Third time's a charm?

phoddie, 2022-01-31:
> Third time's a charm?

Yes!
[image]

Here's the revised patch, which stops on the path separator before the first mismatch:

```js
				const stop = Math.min(first.lastIndexOf("/"), last.lastIndexOf("/"));
				common = 0;
				for (let i = 0; i <= stop; i++) {
					const c = first.charAt(i);
					if (c !== last.charAt(i)) 
						break;

					if ("/" === c)
						common = i;
				}
```

I'm still not 100% confident that it is correct, but it makes more sense.

cmidgley, 2022-01-31:
It works!  I migrated back to my original name, and compiled / tested all of my builds successfully.  

Not sure if you wanted to do any more changes ("I'm still not 100% confident...") so I'm leaving this open, but from my perspective it's closed!  If you do make changes, I'm glad to spin them up for a test.

phoddie, 2022-01-31:
Thanks for confirming that it works. The change may be fine as-is. This makefile stuff is nearly as confusing for me as screen rotation, so I want to give it a little time to settle.

closed 2022-02-09 by phoddie