Moddable-OpenSource/moddable #820: Modules scopes are not respected in the build system

cmidgley, 2022-01-25:
**Build environment:** Linux
**Target device:** ESP32

**Description**
Using module scopes (such as `@myscope/mymodule`) are not correctly respected in the build system.  The scope is removed from the path and all modules are merged into a single directory under `tmp/etc`.

**Steps to Reproduce**
1. Set up a module using a scoped name, and look in the resulting build directory.  While the module names are used for the compiled binaries, all the source files are placed at the same directory without the scope name.

Here is a small fragment of the makefile:

```
$(MODULES_DIR)/@cmidgley/jest-recorder.xsb: $(MODULES_DIR)/jest-recorder/src/main.js
	@echo "# xsc @cmidgley/jest-recorder.xsb"
	$(XSC) $(MODULES_DIR)/jest-recorder/src/main.js -d -c -e -o $(@D) -r jest-recorder
```

Here we can see the source file does not include the `@cmidgley` scope, while the `xsb` file does use the module scope.

phoddie, 2022-01-25:
We will need an complete example to be able to investigate this report. Thank you.

cmidgley, 2022-01-25:
Unable to reproduce with a small setup - It appears it is another example side effect of #819, in this case not including the correct path for the scope.  A clean build has scope, as in this example where `common.js` is included in two scopes, and the two sources files are maintained (correctly) in different directories (unlike #819). 

```
$(MODULES_DIR)/@scope/common.xsb: $(MODULES_DIR)//common.js
	@echo "# xsc @scope/common.xsb"
	$(XSC) $(MODULES_DIR)//common.js -d -c -e -o $(@D) -r common
$(MODULES_DIR)/@alternate/common.xsb: $(MODULES_DIR)//alternate/common.js
	@echo "# xsc @alternate/common.xsb"
	$(XSC) $(MODULES_DIR)//alternate/common.js -d -c -e -o $(@D) -r common
```

closed 2022-01-25 by cmidgley

phoddie, 2022-01-25:
OK. Once #819 is resolved, let's see if this is resolved.