Moddable-OpenSource/moddable #822: esp32/m5paper corrupted screen updates on rotated screen

FWeinb, 2022-01-27:
**Build environment:** macOS
**Target device:** esp32/m5paper

**Description**
When using `screen.rotation = 90` partial screen updates causes the screen to render skewed like this:
[image]

**Steps to Reproduce**
1. Sample code to reproduce this issue:
```js
import {} from 'piu/MC';

export default function () {
  screen.rotation = 90; // This is causing the problem

  screen.configure({ updateMode: 'GLD16' });

  return new Application(null, {
    left: 0, right: 0, top: 0, bottom: 0,
    contents: [
      Label(null, {
        string: 'Test Button',
        skin: new Skin({
          borders: { left: 1, right: 1, top: 1, bottom: 1 },
          fill: ['white', 'black'],
          stroke: 'black',
        }),
        style: new Style({
          margins: { left: 5, right: 5, top: 5, bottom: 5 },
          color: ['black', 'white'],
        }),
        active: true, Behavior: class extends Behavior {
          onTouchBegan(btn) {
            btn.state = 1;
          }
          onTouchEnded(btn) {
            btn.state = 0;
          }
        },
      }),
    ],
    style: new Style({
      font: '35px Open Sans',
      color: 'black',
    }),
    skin: new Skin({ fill: 'white' }),
  });
}
```
2. Build with `mcconfig -d -m -p esp32/m5paper`
3. Touch on the Button

phoddie, 2022-01-27:
Interesting. Maybe I didn't test partial updates when rotated carefully enough. The `adaptInvalid` method of the driver is where those kinds of constraints are applied. I can take a look, but it may be a day or so.

https://github.com/Moddable-OpenSource/moddable/blob/e55fbbf8c9a450acb052fc1630171f04a11c25c2/modules/drivers/it8951/it8951.js#L453-L468

phoddie, 2022-01-28:
Thanks for the test app. That really helps. With thiss revision to `adaptInvalid` below it renders correctly  at 0 and 90:

```js
	adaptInvalid(area) {
		if (area.x & 3) {
			area.w += area.x & 3;
			area.x &= ~3;
		}
		area.w = (area.w + 3) & ~3;

		if (this.#epd._rotation & 1) {
			if (area.y & 3) {
				area.h += area.y & 3;
				area.y &= ~3;
			}
			area.h = (area.h + 3) & ~3;
		}
	}
```

I think that change is consistent with the M5 driver now. Does it work for you too?

If that's correct, the check at the top of `begin` should also be adjusted to match:

```js
		if ((x | width) & 3)
			throw new Error;

		if (this.#epd._rotation & 1) { 
			if ((y | height) & 3)
				throw new Error;
		}
```

FWeinb, 2022-01-28:
That is solving the initial issue. But it looks like the touch input isn't rotated according to the screen rotation.  Updated sample code with the Label moved 200px from the top to reproduce the issue: 
```js
export default function () {
  screen.rotation = 90; 

  screen.configure({ updateMode: 'GC16' });

  return new Application(null, {
    left: 0,
    right: 0,
    top: 0,
    bottom: 0,
    contents: [
      Label(null, {
        top: 200, // Move 200px from the top. You need to touch the lower 
        string: 'Test Button',
        skin: new Skin({
          borders: { left: 1, right: 1, top: 1, bottom: 1 },
          fill: ['white', 'black'],
          stroke: 'black',
        }),
        style: new Style({
          margins: { left: 5, right: 5, top: 5, bottom: 5 },
          color: ['black', 'white'],
        }),
        active: true,
        Behavior: class extends Behavior {
          onTouchBegan(btn) {
            btn.state = 1;
          }
          onTouchEnded(btn) {
            btn.state = 0;
          }
        },
      }),
    ],
    style: new Style({
      font: '35px Open Sans',
      color: 'black',
    }),
    skin: new Skin({ fill: 'white' }),
  });
}
```

phoddie, 2022-01-29:
Nice job tracking that down. Rotation always makes my head spin.

My feeling is that we should respect the behavior of the touch driver as that reflects the behavior of several devices. So, the right fix would be to swap the mapping of 90 and 270 in the it8951 driver.  What do you think?

FWeinb, 2022-01-29:
Thanks, thinking about the rotation and making assumptions is really tricky. From the PIU framework perspective I think having a unified way rotation works feels more appropriate to me. That way you will always now what `90°` is doing without thinking about the implementation details of the underlying driver you are using.  

Also swapping the mapping of 90 and 270  is a little more complicated. The `setup/piu.js` code is extending the `config.Screen`. In this setter function: 
https://github.com/Moddable-OpenSource/moddable/blob/e55fbbf8c9a450acb052fc1630171f04a11c25c2/build/devices/esp32/setup/piu.js#L146-L151
the `rotate` function is determined by the value passed to the setter, this behaviour can't be changed inside the it8951 driver. I might be missing something though.

phoddie, 2022-01-29:
Agreed, a consistent Piu rotation behavior is better for developers building apps. This change to the IT8951 driver code seems to work with all four orientations using your test:

```js
	set rotation(value) {
		const rotation = Math.idiv(value, 90);
		if (rotation & ~3)
			throw new Error;
		this.#epd._rotation = (rotation & 1) ? ((~rotation & 2) | 1) : rotation;
	}
	get rotation() {
		const rotation = this.#epd._rotation; 
		return 90 * ((rotation & 1) ? ((~rotation & 2) | 1) : rotation);
	}
```

FWeinb, 2022-01-29:
Can confirm. This is working in my application too.

phoddie, 2022-01-29:
Thanks for the quick check.

closed 2022-02-09 by phoddie