Moddable-OpenSource/moddable #823: esp32/m5paper wrong battery voltage calculation

FWeinb, 2022-01-29:
**Build environment:** macOS
**Target device:** esp32/m5paper

**Description**
Voltage read by the `device.peripheral.battery.Default` class are incorrect.  

**Steps to Reproduce**
1. Compile this sample code: 
```js
const battery = new globalThis.device.peripheral.battery.Default();
trace(`Voltage is: ${battery.read()}\n`);
```
2. See a value of ~`10,xxxx` being reported for a nearly fully charged battery. 

**Other information**
It looks like this is the problem: 
https://github.com/Moddable-OpenSource/moddable/blob/e55fbbf8c9a450acb052fc1630171f04a11c25c2/build/devices/esp32/targets/m5paper/host/provider.js#L205-L209

I guess this comes from the m5ink implementation but the factory implementations are different. See 
https://github.com/m5stack/M5EPD/blob/bf4bd28f690d30a01310c8f538761f3b070b77f3/src/M5EPD.cpp#L107-L118 and 
https://github.com/m5stack/M5-CoreInk/blob/47951a3a8787e0fbaa999e06ec2d0b55a6583af7/examples/FactoryTest/FactoryTest.ino#L52-L62
respectively. 

After updating the implementation to
```js
let value = this.#analog.read() * 3300;
return value / (1 << this.#analog.resolution) / 0.5;
```
the values look much more reasonable. On thing to not though is that the M5EPD driver is [initialising the ADC_UNIT_1 with a different reference voltage](https://github.com/m5stack/M5EPD/blob/bf4bd28f690d30a01310c8f538761f3b070b77f3/src/M5EPD.cpp#L93) (3600 instead of [V_REF 1100 in the esp32 io module](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/io/analog/esp32/_analog.c#L149)).

phoddie, 2022-01-29:
I never took the time to understand why the battery code borrowed from the M5 repo didn't work. That's why it isn't used in an of the examples. Thank you for taking the time to track that down. Would you prefer to open a PR for that or have me apply your change?

We probably should provide a way to specify VREF when instantiating the constructor. That's a bit bigger project -- to try to do that in a way that might possible be useful for other MCUs. Tagging @andycarle here for reference when we get to that.

FWeinb, 2022-01-29:
Ups... Create a PR and I made a little mess here. Had an old fork of the repository and got confused for a bit.

closed 2022-01-31 by mkellner