Moddable-OpenSource/moddable #826: xsc heap corruption when compiling invalid javascript

FWeinb, 2022-01-30:
**Build environment:** macOS

**Description**
When building invalid javascript the `xsc`  is crashing with a heap corruption. 

**Steps to Reproduce**
1. Create test file `echo "class { constructor() }" > crash.js`
2. Compile with `$MODDABLE/build/bin/mac/debug/xsc crash.js` (happens in release build too)
3. You get following error:
```
[...]/crash.js:2: error: invalid !
[...]/crash.js:2: error: missing identifier!
[...]/crash.js:2: error: invalid !
[...]/crash.js:2: error: missing identifier!
[...]/crash.js:2: error: invalid !
[...]/crash.js:2: error: missing identifier!

[...repeated...]

xsc(92543,0x111b2a600) malloc: Heap corruption detected, free list is damaged at 0x600003dc8030
*** Incorrect guard value: 4375928256
xsc(92543,0x111b2a600) malloc: *** set a breakpoint in malloc_error_break to debug
zsh: abort      $MODDABLE/build/bin/mac/debug/xsc crash.js
```

phoddie, 2022-01-30:
This is a known issue, the result of an intentional behavior to help during development.

`xsc` uses the JavaScript parser in XS to generate the byte code file it outputs. The parser has two different error handling behaviors -- stop on first error or keep parsing. Stopping on first error is safest, and this is what is used when parsing at runtime (e.g. `eval` on an ESP8266 and `xst` both stop on the first error). `xsc` uses the other behavior, because it can be useful to see multiple errors. This attempt to be helpful lis imperfect in two ways: occasionally it generates a long report of redundant errors and sometimes the parser gets into unexpected states which leads to a crash.

It is easy to change `xsc` to stop on the first error, change this line:

https://github.com/Moddable-OpenSource/moddable/blob/e55fbbf8c9a450acb052fc1630171f04a11c25c2/xs/tools/xsc.c#L307

To:

```c
fxInitializeParser(parser, (void  *)-1, 1024*1024, 1993);
```

Perhaps we should change `xsc` to use the safer behavior until the occasional issues are resolved.

phoddie, 2022-02-03:
xsc now stops on the first error, which avoids triggering this failure

closed 2022-02-09 by phoddie