Moddable-OpenSource/moddable #828: TLS Certificate is expired.

stc1988, 2022-01-31:
**Build environment:** macOS, Windows
**Target device:**  desktop simulator

**Steps to Reproduce**
1. Build and Run `examples/network/http/httpsgetjson` using this build command: `mcconfig  -d -m`
2.  TLS Error is thrown at [ssl_handshake.js](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/crypt/ssl/ssl_handshake.js#L504) .
And Console log `date validation failed on certificate resource` is shown. 

TLS Certificates  located in the [`module/ecrypt/data folder`](https://github.com/Moddable-OpenSource/moddable/tree/public/modules/crypt/data) seems to expired.

andycarle, 2022-02-04:
Thank you for the report! We've confirmed the issue and are contemplating the best options to maintain those certificates on our end.

wilberforce, 2022-02-05:
Is there a reason they are named as numeric names- could they be named more descriptively so the filename reflects the cert?

Perhaps symbolic links to the old names?

phoddie, 2022-02-11:
The numbers reflect their position in the `ca.ski` index file. Giving them a friendly name at runtime would require additional data.

stc1988, 2022-06-26:
Now I ran `examples/network/http/httpsgetjson` and got a erorr `Resource: Resource not found: ca222.der!`.
So I added `ca222.der` which is not expired to `manifest.json`. 　Eventually Application ran without any error.

I don't know why this issue solved,  but can we close this issue?

phoddie, 2022-07-01:
Interesting. I just tried the example here. The site is still requesting ca109. Maybe you are somehow reaching a different endpoint?

phoddie, 2022-08-11:
This [note](https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/) from Let's Encrypt seems relevant as the failure is with ca109.der which is the expired `DST Root CA X3`.

phoddie, 2022-08-15:
I rebuilt the certificate bundle with an updated certificate. It is working reliably now. Please update give it a try to confirm. Thank you!

stc1988, 2022-08-17:
Thank you for updating.
This issues seems to have been solved.

closed 2022-08-17 by stc1988