Moddable-OpenSource/moddable #829: Problems erasing flash for ESP32

costa-victor, 2022-01-31:
## **Environment**
**Build environment:** Linux Mint 20 x86_64
**Target device:** NodeMCU ESP32
**Moddable version**: public - commit [e55fbbf](https://github.com/costa-victor/moddable/commit/e55fbbf8c9a450acb052fc1630171f04a11c25c2)
**ESPIDF version**:  [v4.3.2](https://github.com/espressif/esp-idf/tree/v4.3.2)

## **Description**
The **erase_flash** option in mcconfig doesn't work for ESP32.

```bash
victor@victor-ac:~/moddable/examples/io/digital/blink$ mcconfig -i -m -p esp32/nodemcu -t erase_flash
esptool.py v3.2-dev
usage: esptool [-h] [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2}] [--port PORT] [--baud BAUD]
               [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}] [--after {hard_reset,soft_reset,no_reset,no_reset_stub}] [--no-stub] [--trace]
               [--override-vddsdio [{1.8V,1.9V,OFF}]] [--connect-attempts CONNECT_ATTEMPTS]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
               ...

esptool.py v3.2-dev - ESP8266 ROM Bootloader Utility

positional arguments:
  {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
                        Run esptool {command} -h for additional help
    load_ram            Download an image to RAM and execute
    dump_mem            Dump arbitrary memory to disk
    read_mem            Read arbitrary memory location
    write_mem           Read-modify-write to arbitrary memory location
    write_flash         Write a binary blob to flash
    run                 Run application code in flash
    image_info          Dump headers from an application image
    make_image          Create an application image from binary files
    elf2image           Create an application image from ELF file
    read_mac            Read MAC address from OTP ROM
    chip_id             Read Chip ID from OTP ROM
    flash_id            Read SPI flash manufacturer and device ID
    read_flash_status   Read SPI flash status register
    write_flash_status  Write SPI flash status register
    read_flash          Read SPI flash content
    verify_flash        Verify a binary blob against flash
    erase_flash         Perform Chip Erase on SPI flash
    erase_region        Erase a region of the flash
    merge_bin           Merge multiple raw binary files into a single file for later flashing
    version             Print esptool version
    get_security_info   Get some security-related data

optional arguments:
  -h, --help            show this help message and exit
  --chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2}, -c {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2}
                        Target chip type
  --port PORT, -p PORT  Serial port device
  --baud BAUD, -b BAUD  Serial port baud rate used when flashing/reading
  --before {default_reset,usb_reset,no_reset,no_reset_no_sync}
                        What to do before connecting to the chip
  --after {hard_reset,soft_reset,no_reset,no_reset_stub}, -a {hard_reset,soft_reset,no_reset,no_reset_stub}
                        What to do after esptool.py is finished
  --no-stub             Disable launching the flasher stub, only talk to ROM bootloader. Some features will not be available.
  --trace, -t           Enable trace-level output of esptool.py interactions.
  --override-vddsdio [{1.8V,1.9V,OFF}]
                        Override ESP32 VDDSDIO internal voltage regulator (use with care)
  --connect-attempts CONNECT_ATTEMPTS
                        Number of attempts to connect, negative or 0 for infinite. Default: 7.
make: *** [/home/victor/moddable/build/tmp/esp32/nodemcu/instrument/blink/makefile:629: erase_flash] Erro 1
```

## **Steps to Reproduce**
1. In the **examples/io/digital/blink** folder run `mcconfig -i -m -p esp32/nodemcu`
2. Run `mcconfig -i -m -p esp32/nodemcu -t erase_flash`

Also tested for ESP8266 and it's working, but it's a slightly different option: `-t erase`.

## **Expected behavior**
Complete the erase flash memory process.

```bash
victor@victor-ac:~/moddable/examples/io/digital/blink$ mcconfig -i -m -p esp32/nodemcu -t erase_flash
esptool.py v3.2-dev
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: f0:08:d1:c7:8a:38
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 6.8s
Hard resetting via RTS pin...
```

## **More information**
Looking further at the **make.esp32.mk** file, the code which define the **UPLOAD_PORT** variable is empty, then send a invalid syntax for esptool in **port** option (e.g. ```esptool.py --chip esp32 --port erase_flash```).
```makefile
. . .

ifeq ($(UPLOAD_PORT),)
	PORT_SET =
	SERIAL2XSBUG_PORT = $$PORT_USED
else
	PORT_SET = -p $(UPLOAD_PORT)
	SERIAL2XSBUG_PORT = $(UPLOAD_PORT)

. . .

erase_flash:
	$(ESPTOOL) --chip $(ESP32_SUBCLASS) --port $(UPLOAD_PORT) erase_flash

. . .
```

phoddie, 2022-01-31:
This is not a documented feature. It is also obsolete and the vestiges will be removed. Please invoke `esptool` tool directly.

closed 2022-01-31 by phoddie

costa-victor, 2022-02-06:
Hi Mr. @phoddie, thanks for the quick feedback.

I realized it's not documented and I tried to use them as it's a nice feature that unfortunately is not fully working. I made a fix for this feature for ESP32 (it's already working for ESP8266) and also documented it. Hope it can be useful.

phoddie, 2022-02-10:
@costa-victor, thank you for the report and PR. That represents significant effort, which is very much appreciated.

I agree that it would be helpful to provide developers with easy access to tools to manage their device, such as erasing flash. The question is which command (or commands) should do that. `mcconfig` and `mcrun` are about building and deploying. I would like them to keep that focus. As you understand, it is already considerable work to maintain those tools for each development host (macOS, Windows, and Linux) and each embedded target device. For those reason, it seems appropriate to remove flash support there.

The Moddable SDK currently does not have an obvious place to move erase flash to. There is a promising new tool to support developers working with the Moddable SDK called [xs-dev](https://github.com/HipsterBrown/xs-dev) by Nick Hehr (@HipsterBrown) that feels like a good candidate. xs-dev automates many common development tasks, including set-up, updating, creating new projects, building and deploying, running examples, etc. The broad scope and extensible architecture would appear to be a good fit for device management capabilities, such as erasing flash. I imagine Nick would be open to discussing the idea, if you are interested in exploring that.

HipsterBrown, 2022-02-10:
Thanks for the mention @phoddie. `xs-dev` is definitely a great place for organizing useful commands related to working with Moddable projects. 

@costa-victor If you can create an issue in the xs-dev repo to describe what the feature could look like, I'm opening to pull requests and/or pairing on the solution!

costa-victor, 2022-02-26:
Thank you Mr. @phoddie for the clarification and for also showed me xs-dev project. I`m glad to hear that @HipsterBrown, I would love to contribute with xs-dev project.