Moddable-OpenSource/moddable #831: Module names with ":" generate incorrect tsconfig.json

cmidgley, 2022-02-01:
**Build environment:** Linux
**Target device:** ESP32

**Description**
When there is a colon in the name of a module, the generated `tsconfig.json` uses an incorrect name for the module.  For example, in `manifest.json` specify:

```json
"modules": {
   "embedded:io/*": ["io/*"]
}
```

The resulting generated `tsconfig.json` looks like this (when matching against file `io/digital.d.ts`):

```json
"~.embedded/io/digital": [
   "/workspace/libraries/util/src/io/digital"
]
```

The correct generation should be:
```json
"embedded:io/digital": [
   "/workspace/libraries/util/src/io/digital"
]
```

I have verified that if I run `sed` against the `tsconfig.json` and fixup the names as mentioned above (after it is generated, and before `tsc` executes), the code will compile correctly and the `embedded:io/digital` module will correctly load.

**Steps to Reproduce**
1. Clone [this repo](https://github.com/cmidgley/moddable-modules-with-colon)
2. From the root, run `mcconfig -m -d -p esp32`
3. Receive error from `tsc`  that `embedded:io/digital` cannot be found
4. Open generated `tsconfig.json` file, and notice `~.embedded/io/digital` is being used.

phoddie, 2022-02-03:
Thanks for the example repository. That made tracking this down (mostly) straightforward.

To handle the namespace prefix, `mcconfig` and `mcrun` have to do some [extra path mapping](https://github.com/Moddable-OpenSource/moddable/blob/a4e0ec573b647af85ea2879e9dc7e22973a3634a/tools/mcmanifest.js#L1719-L1725). When generating the paths section `tsconfig.json` that mapping has to be reversed because the path prefix is also used as the module specifier. There is already code to reverse the slash mapping on Windows:

https://github.com/Moddable-OpenSource/moddable/blob/a4e0ec573b647af85ea2879e9dc7e22973a3634a/tools/mcmanifest.js#L862-L867

To reverse the mapping of the namespace prefix, add two lines between 865 and 866:

```js
			if (specifier.startsWith("~."))
				specifier = specifier.slice(2).replace("/", ":");
```

Please let me know if this works for you too. The combination of makefiles and TypeScript is always tricky. I'd like to let this settle for a day or three before deciding if it is really the correct solution.

phoddie, 2022-02-07:
Great. Thank you for confirming. I've reorganized the change a bit to make the intent a bit clearer. That will be part of our next update.

closed 2022-02-09 by phoddie