Moddable-OpenSource/moddable #833: Enabling SPIRAM Support for m5paper results in no boot

FWeinb, 2022-02-06:
**Build environment:** macOS
**Target device:** m5paper

**Description**
I tried enabling the SPIRAM on the m5paper with no success. As soon as `CONFIG_ESP32_SPIRAM_SUPPORT=y` is enabled in the sdkconfig the esp32 won't boot anymore. The m5paper is using a `ESP32-D0WDQ6-V3` which is configured to use the IO17 (CLK) and IO16 (CS). Looking at the [schematic this looks correct](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/schematic/Core/m5paper/M5_PAPER_SCH.pdf) (see ESPSRAM64H section). I don't really now how to debug this as I don't get any output on Serial anymore (even after increasing the log level to INFO for both the bootloader as the application). I tried disabling the [CONFIG_SPIRAM_BOOT_INIT](https://demo-dijiudu.readthedocs.io/en/latest/api-reference/kconfig.html#spiram-boot-init) and and [SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-allow-stack-external-memory) but nothing worked. 

**Steps to Reproduce**
1. Change this to `y`
https://github.com/Moddable-OpenSource/moddable/blob/a4e0ec573b647af85ea2879e9dc7e22973a3634a/build/devices/esp32/targets/m5paper/sdkconfig/sdkconfig.defaults#L1
2. Build using `mconfig -d 
3. xsbug.app will never connect to the device

**Expected behavior**
It should be possible to use the available SPIRAM.

phoddie, 2022-02-06:
> It should be possible to use the available SPIRAM

Yes, it should. Clearly setting Espressif's basic configuration option is not enough.

Strictly speaking, I don't think this is a Moddable SDK bug. I would be very happy to learn how to solve this though.

FWeinb, 2022-02-07:
Had some time to investigate a little more. I did notice the default console was disabled via the `CONFIG_ESP_CONSOLE_UART_NONE` flag, thats why i wasn't getting any debug output. Enabling the default `CONFIG_ESP_CONSOLE_UART_DEFAULT=y` console I now see that the spi_flash is correctly initialised: 
```
I (22) psram: This chip is ESP32-D0WD
I (26) spiram: Found 64MBit SPI RAM device
I (30) spiram: SPI RAM mode: flash 80m sram 80m
I (34) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
[...]
I (885) spiram: Adding pool of 4096K of external SPI memory to heap allocator
```

But then when trying to start the app via `esp_startup_start_app` the ESP is crashing:
```
D (950) cpu_start: calling init function: 0x401aecc4
0x401aecc4: _GLOBAL__sub_I___cxa_get_globals_fast at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_globals.cc:145

D (955) cpu_start: calling init function: 0x4008e34c
0x4008e34c: esp_reset_reason_init at /Users/fweinb/esp32/esp-idf/components/esp_system/port/soc/esp32/reset_reason.c:73

D (960) cpu_start: calling init function: 0x4015f8c8
0x4015f8c8: s_set_default_wifi_log_level at /Users/fweinb/esp32/esp-idf/components/esp_wifi/src/wifi_init.c:72

D (964) cpu_start: calling init function: 0x40158714
0x40158714: esp_ipc_init at /Users/fweinb/esp32/esp-idf/components/esp_ipc/ipc.c:102

D (969) cpu_start: calling init function: 0x400d0a84
0x400d0a84: esp_ota_init_app_elf_sha256 at /Users/fweinb/esp32/esp-idf/components/app_update/esp_app_desc.c:76

D (974) intr_alloc: Connected src 17 to int 3 (cpu 0)
I (978) esp_core_dump_uart: Init core dump to UART
D (983) efuse: coding scheme 0
D (986) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (992) efuse: coding scheme 0
D (995) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit

assert failed: 0x401d159a
0x401d159a: esp_startup_start_app at /Users/fweinb/esp32/esp-idf/components/freertos/port/xtensa/port.c:534 (discriminator 1)
Backtrace:0x400817ad:0x3ffe3b80 0x4008e3c1:0x3ffe3ba0 0x40094345:0x3ffe3bc0 0x401d159a:0x3ffe3c00 0x400d1123:0x3ffe3c20 0x40081525:0x3ffe3c60 0x400792a3:0x3ffe3c90 |<-CORRUPTED
0x400817ad: panic_abort at /Users/fweinb/esp32/esp-idf/components/esp_system/panic.c:404
0x4008e3c1: esp_system_abort at /Users/fweinb/esp32/esp-idf/components/esp_system/system_api.c:112
0x40094345: __assert_func at /Users/fweinb/esp32/esp-idf/components/newlib/assert.c:47
0x401d159a: esp_startup_start_app at /Users/fweinb/esp32/esp-idf/components/freertos/port/xtensa/port.c:534 (discriminator 1)
0x400d1123: start_cpu0_default at /Users/fweinb/esp32/esp-idf/components/esp_system/startup.c:429
0x40081525: call_start_cpu0 at /Users/fweinb/esp32/esp-idf/components/esp_system/port/cpu_start.c:536
```

and the core dump:
```
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffafae8, name: 'ipc0', GDB name: 'process 1073412840'

================== CURRENT THREAD REGISTERS ===================
exccause       0xffff (InvalidCauseRegister)
excvaddr       0x0
pc             0x20000000          0x20000000
lbeg           0x0                 0
lend           0x0                 0
lcount         0x0                 0
sar            0x0                 0
ps             0x30                48
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
[... 44 lines cut ...]
Name   Address   Size   Attrs
.rtc.text 0x400c0000 0x0 RW
.rtc.dummy 0x3ff80000 0x0 RW
.rtc.force_fast 0x3ff80000 0x0 RW
.rtc.data 0x50000000 0x10 RW A
.rtc_noinit 0x50000010 0x0 RW
.rtc.force_slow 0x50000010 0x0 RW
.iram0.vectors 0x40080000 0x403 R XA
.iram0.text 0x40080404 0x153e7 RWXA
.dram0.data 0x3ffb0000 0x49cc RW A
.noinit 0x3ffb49cc 0x0 RW
.ext_ram.bss 0x3f800000 0x0 RW
.flash.appdesc 0x3f400020 0x100 R  A
.flash.rodata 0x3f400120 0x769a8 RW A
.flash.text 0x400d0020 0x10191f R XA
.iram0.data 0x400957ec 0x0 RW
.iram0.bss 0x400957ec 0x0 RW
.dram0.heap_start 0x3ffb8c00 0x0 RW
.coredump.tasks.data 0x3ffafae8 0x158 RW
.coredump.tasks.data 0x20000000 0x70 RW
.coredump.tasks.data 0x3ffafd4c 0x158 RW
.coredump.tasks.data 0x3ffb9740 0x128 RW
.coredump.tasks.data 0x3ffba874 0x158 RW
.coredump.tasks.data 0x3ffba740 0x12c RW
```

phoddie, 2022-02-07:
> Had some time to investigate a little more. I did notice the default console was disabled via the CONFIG_ESP_CONSOLE_UART_NONE flag, thats why i wasn't getting any debug output

Apologies. I should have mentioned that when you want to see the ESP-IDF output, you can run an instrumented build. That enables the console output and disables xsbug (which won't work reliably with console output from ESP-IDF enabled).

```
mcconfig -i -m -p esp32
```

Congratulations on getting further.  It is still dying pretty early though. Some SPI flash doesn't work reliably at 80 MHz. Maybe the same is true of PSRAM. You could try slowing that down to 20 MHz to see?

FWIW - the Moddable SDK can boot with SPIRAM on ESP-iDF. We have it enabled on the Kaluga (ESP32-S2) target, as I recall.

FWeinb, 2022-02-08:
Thanks for hinting the instrumented build. I got the PSRAM working with enabling `CONFIG_ESP_INT_WDT=y`.

phoddie, 2022-02-08:
>  I got the PSRAM working with enabling CONFIG_ESP_INT_WDT=y...

That's great!

> ...but that breaks xsbug.

Yes, right. I remember that from a while ago. When stopped at a breakpoint, xsbug is waiting in a loop and that would trigger the WDT. But, we don't enable the WDT in a debug build so there's no issue.

I believe that wait loop calls `fxReceive` continuously  so you can just add a call to the watchdog function `esp_task_wdt_reset` [there](https://github.com/Moddable-OpenSource/moddable/blob/4e39861fe5f00b5006393a0b42f84b64f17b5ea0/xs/platforms/esp/xsPlatform.c#L737-L739).  If that works, we will need to sort out the right conditional so that is only included when needed.

FWeinb, 2022-02-08:
I should be more specific, feeding the WDT isn't an issue i guess. I get some strange Serial output when the WDT is enabled:
```
Executing "ninja flash"...
Done
574�X*U�	<;��08��
```

I guess there is some cross talk between the UART and PSRAM.

phoddie, 2022-02-08:
That is strange. I don't really have an idea at the moment. Maybe some of our code or data needs to be forced into main RAM (not PSRAM).

phoddie, 2022-02-10:
Trying to think about this some more:

- Could it be diagnostic output from the PSARM support?
- Is it possible that the PSRAM initialization is somehow messing up the baud rate? The Moddable SDK initialize the baud after PSRAM has been initialized, so that shouldn't be a problem. But...
- If you run an instrumented build, you should see the instrumentation traces once per second from the Moddable SDK. Are those uncorrupted? Do you see any other serial output?

I can try to take a look at this too, but maybe not until the weekend.

FWeinb, 2022-02-10:
Thanks for locking into this. The instrument build didn't give me any output. It looks like [`sdkconfig.inst`](https://github.com/Moddable-OpenSource/moddable/blob/3458406d8c63783019e58d03de9e2a7055f1983c/build/devices/esp32/xsProj-esp32/sdkconfig.inst#L12) is used to overwrite the `CONFIG_ESP_CONSOLE_NONE` to `no` but when I look at the generated `sdkconfig.mc` in the build folder `build/tmp/esp32/instrument/.../xsProject-esp32` the value is still `CONFIG_ESP_CONSOLE_NONE=y` which prevent the console output.
When enabling it via `idf.py menuconfig` i get the following output though:
```
[...]
instruments: 0,0,0,0,0,2,0,0,0,0,4340310,8,48,11276,11352,27360,27648,3792,5120,38,6,0,0,60805
instruments: 0,0,0,0,0,2,0,0,0,0,4340310,0,100,11312,11352,25168,27648,1648,5120,1,6,0,0,0
instruments: 0,0,0,0,0,2,0,0,0,0,4340310,0,100,11348,11352,25504,27648,1648,5120,0,6,0,0,0
```

Based on [this array in instruments](https://github.com/Moddable-OpenSource/moddable/blob/658705257dfec22cef948d160bd87c259b303fdc/examples/base/instrumentation/instrumentation/main.js#L17) i guess the PSRAM is working correctly as `4340310` indicates more then 4 MB of RAM available.

Why this isn't working in the debug build I could not figure out yet.

phoddie, 2022-02-10:
Yes, that's the expected output. That's great -- especially the 4 MB of free RAM! (Not sure why the generated SDKCONFIG is wrong, but let's stick to one mystery at a time).

> Why this isn't working in the debug build I could not figure out yet.

The debug build runs the serial port at 460800 baud (for fast debugging), whereas the release & instrumented builds are at 115200 (for compatibility with ESP32 tools). Maybe slowing down the debug build baud rate would help? You can set `DEBUGGER_SPEED` in the environment to do that, or just edit the [make file](https://github.com/Moddable-OpenSource/moddable/blob/3458406d8c63783019e58d03de9e2a7055f1983c/tools/mcconfig/make.esp32.mk#L23).

FWeinb, 2022-02-10:
> especially the 4 MB of free RAM! 
I am still getting `XS abort: memory full` so might not be fully working yet. I have the `CONFIG_SPIRAM_USE_MALLOC=y` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y` enabled. 

> You can set DEBUGGER_SPEED in the environment to do that, or just edit the [make file](https://github.com/Moddable-OpenSource/moddable/blob/3458406d8c63783019e58d03de9e2a7055f1983c/tools/mcconfig/make.esp32.mk#L23).
I must be doing something wrong as neither doing `DEBUGGER_SPEED=115200 mcconfig -d -m -p` nor changing the file seams to change that value. When doing a verbose build I still get `-D DEBUGGER_SPEED=921600` there.

phoddie, 2022-02-10:
Is the "memory full" from a debug build?

If you can let me know the changes you've made, I can try to take a look for a few minutes tonight. Maybe I'll notice something.

phoddie, 2022-02-11:
Good news. Using your SDKCONFIG xsbug works, if the baud rate is turned down to 115200! Have a look:

[image]

When I tried a debug build at 460800 and 230400 baud, the serial stream from the M5Paper was corrupt as you observed.

I think the reason it didn't work for you may be that xsProj-esp32/main/main.c didn't get rebuilt. That file contains the [code](https://github.com/Moddable-OpenSource/moddable/blob/3458406d8c63783019e58d03de9e2a7055f1983c/build/devices/esp32/xsProj-esp32/main/main.c#L228-L242) that initializes the serial port. 

Your theory about cross-talk is looking pretty good. Maybe the M5Paper hardware design doesn't support high speed serial when PSRAM is enabled. That's kind of obscure, so perhaps it wasn't something tested for. Or maybe there is some configuration option we are overlooking?

phoddie, 2022-02-11:
Note that you can specify `DEBUGGER_SPEED` in the manifest. If we enable PSRAM by default for M5Stack, that will need to be configured:

```
	"build": {
		"DEBUGGER_SPEED": "115200"
	},
```

phoddie, 2022-02-11:
I didn't see a crash running epaper-flashcards. What did I miss?

FWeinb, 2022-02-11:
Oh, sorry, the epaper-flashcards example isn't running out of memory. I am using my app shell that is crashing when doing a WiFi-Scan and trying to display a list of found networks. Need to think about a test case for this crash.

phoddie, 2022-02-11:
No problem. Just let me know how to reproduce the problem and I can try to take a look. Otherwise, it seems like maybe the PSRAM support is working.

FWeinb, 2022-02-12:
I did a very simple test filling up an array with values while watching the memory consumption und don't quite understand the numbers that are presented to me.  [Here is a gist of the code](https://gist.github.com/FWeinb/8cc0900868166a85554c0886b97fddb3)

I also made a video showing the crash: 

https://user-images.githubusercontent.com/1250430/153708653-fbd56c98-744e-4160-b120-7a3f9498ffa9.mov

 

Do i need to increate the memory the VM is getting? Is there a way to allow it to allocate memory dynamically?

FWeinb, 2022-02-12:
I have some more insights now. I looks like the stack of a task created with `xTaskCreate` will always be placed in internal ram, not in PRAM. For that you would have to set `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y` and use `xTaskCreateStatic` providing a chunk of memory in PRAM (See [Restrictions](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#restrictions)).

phoddie, 2022-02-12:
Thank you for the examples. That makes it very clear what you are seeing.

What you are seeing is expected. The allocator for the VM on embedded performs a static allocation when the VM is created (that is the value you can change in the manifest). It doesn't grow from there. You can use a dynamic allocator instead. Because there is no MMU, using dynamic allocation means that the XS heap fragments as it grows, so it is a less efficient use of memory. If you want to try that, I can provide some suggestions.

Regarding the stack of the FreeRTOS task, that is usually pretty small (8 or 10 KB). Unless you have many workers (and consequently many native stacks), it is probably best in internal RAM so it can always be fast.

FWeinb, 2022-02-12:
Thanks for the explanation, that makes total sense saw a few places where you do `If (NULL == the->heap)` checks and do `c_malloc` there and got confused. Having this memory statically allocated also ensures that it is definitely placed in PRAM.

Do you want me to prepare a PR to add support for PSRAM, including the `DEBUGGER_SPEED` reduction for the m5paper?

phoddie, 2022-02-12:
> Thanks for the explanation, that makes total sense saw a few places where you do If (NULL == the->heap) checks and do c_malloc there and got confused. Having this memory statically allocated also ensures that it is definitely placed in PRAM.

The `c_malloc` left over from earlier revisions. I should clear that out, as it is misleading. I don't think full dynamic allocation is usually the right choice on embedded as a script can start to use space to the point where system functions start to fail. Supporting dynamic allocation on the device is straightforward (at least for those with dynamic memory allocation!) if that is ever the right choice for a project.

> Do you want me to prepare a PR to add support for PSRAM, including the DEBUGGER_SPEED reduction for the m5paper?

That would be great, thank you.

phoddie, 2022-02-15:
Closing, as the PR to implement this has been integrated.

closed 2022-02-15 by phoddie