Moddable-OpenSource/moddable #835: xsbug crash

mauroForlimpopoli, 2022-02-07:
Good morning,
suddenly my xsbug stopped to run. 
During the writing code it appears with the hourglass (for about 15 seconds) and then close the form.
After that this is the message on the  command promt:
"
  Done
          1 file copiati.
          1 file copiati.
          1 file copiati.
          1 file copiati.
  Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
  C:\moddable\tools\serial2xsbug\serial2xsbug_win.c(107): Impossibile stabilire la connessione. Rifiuto persistente del computer di destinazione.
  NMAKE : fatal error U1077: 'C:\moddable\build\bin\win\release\serial2xsbug.EXE' : codice restituito '0x1'
  Stop.
"
I tried to lanch it directly from the file manager and I got the same result.
This the the sceenshot
How can I resolve ?
regards

mauroForlimpopoli, 2022-02-07:
[image]

phoddie, 2022-02-07:
From the limited information provided, it doesn't look like xsbug crashed. The error shown is the `serial2xsbug` bridge failing when it attempts to connect to xsbug.

mauroForlimpopoli, 2022-02-07:
The Moddable Two is correctly programmed

andycarle, 2022-02-07:
Hi @mauroForlimpopoli,

Just to be sure I understand the issue you're seeing: is `xsbug` hanging like this when you run it directly from the Start Menu or File Explorer, even when no `mcconfig` / `serial2xsbug` session is running? 

If that's the case, a reasonable next step would be to try clearing the `xsbug` preferences in the Windows Registry. Please proceed with caution if you're not experienced using the Registry Editor, but the steps are:

1. Open Registry Editor from the Start Menu.
2. Search the registry (Edit->Find...) for `moddable.tech` with only "Look at: Keys" selected, as below.

[image]

This should find a key of the form `Computer\HKEY_USERS\?????\SOFTWARE\moddable.tech` where the "?????" will be your user account's [security identifier](https://en.wikipedia.org/wiki/Security_Identifier). 

This registry key will include subkeys for each Moddable SDK application that has run on Windows and saved preferences. For example, mine currently has `mcsim`, `Screen Test`, `tech.moddable.timecard`, and `xsbug`.

[image]

3. With `xsbug` closed, delete the `xsbug` registry key and its subkeys. 
4. Re-launch `xsbug` from File Explorer. It will now launch with no open files and no saved preferences, hopefully clearing up whatever is causing your hang.

Please give that a shot and let me know if it helps.

&nbsp;- Andy

mauroForlimpopoli, 2022-02-07:
Dear Andy,
in the registry key "mcsim" is not present. 
do i have to do this test anyway?
Thank you for your assistance.
Mauro

mauroForlimpopoli, 2022-02-07:
P.S: this is the address of the key:
Computer\HKEY_CURRENT_USER\SOFTWARE\moddable.tech\xsbug\window

andycarle, 2022-02-08:
Just delete the key `Computer\HKEY_CURRENT_USER\SOFTWARE\moddable.tech\xsbug`. It should probably have a subkey called `window` and one called `preferences`, as shown above.

mauroForlimpopoli, 2022-02-08:
I did that . Now it works but it shows only the "log" and the "instruments", no file is possible to open and to fix a breakpoint . The same if it starts from the file manager.
May I do something other ?

andycarle, 2022-02-08:
Great! I'm glad to hear that that fixed the hang you were experiencing. 

To set a breakpoint in a file, just navigate to that file in File Explorer and then drag-and-drop the file into the code view pane in xsbug.

mauroForlimpopoli, 2022-02-09:
Unfortunately is not  possible do drag and drop the file into the code view of xsbug. both whien the programming is running or not..

andycarle, 2022-02-09:
Curious. What happens when you try? It should look like this:

[image]

mauroForlimpopoli, 2022-02-09:
It appears a hourglass and after 15 seconds it crashs

andycarle, 2022-02-09:
Okay, well, whatever is causing that might also be what was causing your original hang. I take it this used to work correctly... so, did anything recently change about the location of the code you're trying to debug? Is it in a path with an unusual character or on a network-mounted share? Anything usual about the location in any way you can think of?

mauroForlimpopoli, 2022-02-09:
I tried to drag other files and it works. It could be because the file size if above 245 kb?

mauroForlimpopoli, 2022-02-09:
I cutted just a few line of code and now it seems works. I will split my file accurately and I will test it again

mauroForlimpopoli, 2022-02-09:
It seems that if the file to load is over 247 kb the xsbug can't load it. Let me know. 
Thank you Andy  your precious time

andycarle, 2022-02-09:
Interesting! We'll take a look at that; thank you for the report.

andycarle, 2022-02-09:
@mauroForlimpopoli Is the script file that is causing you problems something that you could share with us for testing? You could email it to me if sharing publicly is problematic.

mauroForlimpopoli, 2022-02-09:
Dear Andy, 

I can't share publicly all my project , but I send it to you. Send me your mail 

Thank you

PS: the code now is very chaotic but it works :-)

Mauro Lorenzi

andycarle, 2022-02-09:
Thanks! I'm andy@moddable.com

andycarle, 2022-02-14:
@mauroForlimpopoli Thank you again for this report. We have a fix for this issue that will go live with our next Moddable SDK update.

andycarle, 2022-02-15:
@mauroForlimpopoli The fix for this is live with commit dfdfeef. When you get a chance, please rebuild the Moddable SDK tools and then see if this corrects your xsbug crash.

Thanks!

mauroForlimpopoli, 2022-02-20:
Dear Andy,

I did the modification.

How can I rebuild the Moddable SDK tools ?

Just another question:

the documents about xbug said If the "modules" are colored "Blue" it means the module are preloade in the flash. What mean "Modules Loaded" on the real time view that is (in my project) always Zero ?

Thank you

Mauro Lorenzi

phoddie, 2022-02-20:
> How can I rebuild the Moddable SDK tools ?

The Getting Started document in the Moddable SDK contains a section on updating for each development platform, including [Windows](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#updating-1). Steps 2 and 3 ensure that you have an up-to-date build of the Moddable SDK tools.

mauroForlimpopoli, 2022-02-20:
Thank you both,
I did it and it seems to work perfectly
Great job

closed 2022-02-20 by phoddie