Moddable-OpenSource/moddable #843: Error with TLS session.alert.level=2 session.alert.description=70

eltoroit, 2022-02-12:
Hi,

I am trying to get some JSON data from here: https://lwcdata.herokuapp.com/jsonData using this code:

```
import { Request } from "http";
import SecureSocket from "securesocket";

let request = new Request({
	host: "lwcdata.herokuapp.com",
	path: "/jsonData",
	response: String,
	Socket: SecureSocket,
	port: 443
});

request.callback = function (stage, value) {
	if (Request.responseComplete == stage)
		trace(value, "\n");
	else if (Request.error == stage) {
		trace(value, "\n");
	}
}
```

But it's throwing an exception and the callback's stage is the `error` with this value: `TLS: alert: 2, 70`

Looking [here](https://techcommunity.microsoft.com/t5/iis-support-blog/ssl-tls-alert-protocol-and-the-alert-codes/ba-p/377132), I can see the error 70 refers to a `protocol_version: The protocol version the client attempted to negotiate is recognized, but not supported. For example, old protocol versions might be avoided for security reasons. This message is always fatal.`

Is there something I am not doing in my code?

eltoroit, 2022-02-12:
Does Moddable support TLS 1.2? Is that the issue?

eltoroit, 2022-02-12:
I found the problem. I need to use TLS 1.2 and Moddable supports it. I need to change the code to this:

```
let request = new Request({
	host: "lwcdata.herokuapp.com",
	path: "/jsonData",
	response: String,
	Socket: SecureSocket,
	port: 443,
	secure: { protocolVersion: 0x303 }
});
```

It's documented here: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/securesocket.md#constructordictionary

phoddie, 2022-02-13:
Yes, that's correct. Congrats on figuring it out. Thanks for posting the solution.

closed 2022-02-13 by phoddie