Moddable-OpenSource/moddable #844: Moddable SDK should support ESP32 built-in Ethernet

JoshuaPK, 2022-02-12:
**Describe**

The ESP32 microcontrollers have built-in Ethernet capability, only requiring a transceiver and handful of discrete components to connect it to a port.  There are more hobbyist-level boards coming out with ESP32 microcontrollers and Ethernet transceivers that connect right to the ESP32 itself (instead of through a controller such as ENC28J60).  Some of these boards are excellent choices for remote monitoring because they have built-in PoE power supplies:

http://www.lilygo.cn/claprod_view.aspx?TypeId=21&Id=1344&FId=t28:21:28
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware

Also, OEMS who design products may go with the built-in Ethernet to save component costs.

JoshuaPK, 2022-02-13:
Thanks @phoddie !  I know that the SDK supports the ENC28J60.  What I am looking for is support for the Ethernet that is built-in to the ESP32 itself.

phoddie, 2022-02-13:
Thanks for the clarification. I was reading too quickly.

The Moddable SDK Ethernet support is largely independent of the hardware used to implement Ethernet. The Ethernet driver is [set-up in `initEthernet`](https://github.com/Moddable-OpenSource/moddable/blob/3458406d8c63783019e58d03de9e2a7055f1983c/modules/network/ethernet/esp32/ethernet.c#L226-L254). That would need to be modified to work with the driver for the built-in Ethernet instead of an external controller.

andycarle, 2023-02-01:
The Moddable SDK [now supports](https://github.com/Moddable-OpenSource/moddable/commit/b8bcd255ee537b3ba1093a6ae660eb7e54a6caa1) the ESP32's built-in Ethernet MAC. This support has been tested with the [WT32-ETH01](http://www.wireless-tag.com/portfolio/wt32-eth01/) development board, which uses the `LAN87XX` external PHY. 

The other external PHYs supported by the `esp_eth` ESP-IDF component can also be used with an appropriate `manifest.json` define. See the [WT32-ETH01 manifest](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/wt32_eth01/manifest.json) for an example and [ethernet.c](https://github.com/Moddable-OpenSource/moddable/blob/819c387912fe0d8b53582b27c5739f1c261c3688/modules/network/ethernet/esp32/ethernet.c#L253) for the appropriate defines. 

Please let me know if you have any issues getting this working on your target development board.

closed 2023-02-01 by andycarle