Moddable-OpenSource/moddable #845: Question on the HTTP Server sample code

eltoroit, 2022-02-13:
[The sample code for the HTTP Get server](https://github.com/Moddable-OpenSource/moddable/blob/3458406d8c63783019e58d03de9e2a7055f1983c/examples/network/http/httpserver/main.js#L21) has this assignment `this.path = value;`

What happens with multiple concurrent requests? Is `this` the server or each independent request?

eltoroit, 2022-02-13:
Never mind, I find my answer [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/network.md#callbackmessage-val1-val2-1)

> A new HTTP Request is instantiated for each incoming request. The callback is invoked with `this` set to the callback instance for the request. The callback function may attach properties related to handling a specific request to this, rather than using global variables, to ensure there are no state collisions when there are multiple active requests.

eltoroit, 2022-02-13:
Thanks for building this awesome SDK!

phoddie, 2022-02-13:
Thanks for sharing the answer to your question.

closed 2022-02-13 by phoddie