Moddable-OpenSource/moddable #847: Difficult to build MacOS version

fernandotetu, 2022-02-16:
**Build environment**
-  MacBook Pro (15-inch, 2017)
-  Intel Core i7
- macOS Monterey (12.2.1) 
- Xcode 13.2.1 (Build version 13C100)

**Description**
I am trying to install Moddable SDK on Mac following the get started guide from the master. However, when I used the make command, I had an error to build a simulator::

	cd ${MODDABLE}/build/makefiles/mac
	make

logs:

	...
	make -f simulator.mk
	# simulator debug : cc main.m
	# simulator debug : cc simulator
	make[1]: *** [/Users/fernando/moddable/build/bin/mac/debug/Screen Test.app/Contents/Resources/English.lproj/main.nib] 	Error 255
	make: *** [debug] Error 2

If I edit the file "${MODDABLE}/build/makefiles/mac/simulator.mk"  and change line 67 in order to remove the main.nib entry, and I am able to proceed with the build process. But, it does not work properly. 

	simulator.mk Line 67
	build: $(TMP_DIR) $(APP_DIR) $(APP_DIR)/main $(RES_DIR)/English.lproj/main.nib
	changed
	67 build: $(TMP_DIR) $(APP_DIR) $(APP_DIR)/main 

From this "adapted version," I Could not get logs on xsBug when running on MAC platform, but it works on ESP32.
I  believe that situation is related to the version of MacOS because I already had success in building Moddable SDK on the previous SO version. 

**Other information**
 I built a new version for Linux (in another laptop with Fedora 29), and it works fine.

phoddie, 2022-02-16:
Using a clean pull of the Moddable SDK from GitHub, I'm not able to reproduce this error. The failure seems to be because a resource file cannot be found. Let's check what you have. Use "Show Package Contents" in Finder to inspect the contents of `$MODDABLE/build/bin/ma/debug/Screen Test.app`. This image shows what I see. Is something different for your build?

[image]

fernandotetu, 2022-02-16:
Hi @phoddie , thank you for you assistance.

In fact, this file does not exist on my machine. Also, I can't find it in the repository. So, is it generated during the build process? 

I put an empty file inside this repository and tried again. The making process progressed. However, it finished with an error (the same error).

I think this issue has started after I installed the pico SDK because it required a new version of the Xcode command-line tool, but I am sure about it. Maybe the CMake can change the C compiler or any configuration. I suppose it is because I get some deprecated warning on the build process, for example:

	...
	# cc pcWorker.c.o
	/Users/fernando/moddable/modules/piu/PC/mac/piuDialogs.m:116:15: warning: 	'setAllowedFileTypes:' is deprecated: first deprecated in macOS 12.0 - Use 	-allowedContentTypes instead [-Wdeprecated-declarations]
                        [openPanel setAllowedFileTypes: [NSArray arrayWithObject:[NSString stringWithUTF8String:string]]];
                                   ^
	/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/AppKit.framework/Headers/NSSavePanel.h:215:49: note: property 'allowedFileTypes' is declared deprecated here
	@property (nullable, copy) NSArray<NSString *> *allowedFileTypes API_DEPRECATED("Use -allowedContentTypes instead", macos(10.3,12.0));
                                                ^
	/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/AppKit.framework/Headers	/NSSavePanel.h:215:49: note: 'setAllowedFileTypes:' has been explicitly marked deprecated here
	1 warning generated.
	/Users/fernando/moddable/modules/piu/PC/Screen/mac/piuScreen.m:659:75: warning: 'kUTTypePNG' is deprecated: first deprecated in macOS 12.0 - Use UTTypePNG instead. [-Wdeprecated-declarations]
	CGImageDestinationRef destination = CGImageDestinationCreateWithURL(url, kUTTypePNG, 1, NULL);
                                                                                 ^
	/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks	/LaunchServices.framework/Headers/UTCoreTypes.h:729:26: note: 'kUTTypePNG' has been explicitly marked deprecated here
	extern const CFStringRef kUTTypePNG                                  API_DEPRECATED("Use UTTypePNG instead.", ios(3.0, 15.0), macos(10.4, 12.0), tvos(9.0, 15.0), watchos(1.0, 8.0));
                         ^
	1 warning generated.
	...

If you need more detail or information, please let me know. Once again, thank you very much.

phoddie, 2022-02-16:
> In fact, this file does not exist on my machine

By "this file, I understand that you mean `$MODDABLE/build/bin/ma/debug/Screen Test.app`.

The warnings from building pcWorker.c are expected. They are distracting and should be cleaned up, but they aren't causing the build failure.

Perhaps you had a build get interrupted and that left the build results in a bad state? Just to be sure, you could try a clean build Just manually delete `$MODDABLE/build/bin/mac` and `$MODDABLE/build/tmp/mac` and then rebuild as you did above.

fernandotetu, 2022-02-17:
Hi, **I can fix it!!!**  

> By "this file, I understand that you mean $MODDABLE/build/bin/ma/debug/Screen Test.app.

I'm sorry for not be clear, I don't find the resource (English.lproj/main.nib). 

Regarding building, In my tests, I always remove Moddable's folder and clone a new instance of the repository.

**How to fix**  

- Remove the Moddable's preferences file from `~/Library/Preferences`
  tech.moddable.xsbug.plist   and tech.moddable.screen-test.plist  
- Select full Xcode installation using the command: 
`sudo xcode-select -s /Applications/Xcode.app/Contents/Developer`

After these steps, I could build the solutions as expected. Thank you so much for your help and assistance

closed 2022-02-17 by fernandotetu

phoddie, 2022-02-17:
Congratulations on getting your build working. It is still a mystery why it failed. 

FWIW – macOS caches preferences in memory, so manually deleting the preference files doesn't always work as expected. The most reliable way we've found to remove preferences is using the `defaults` command. For example this resets the xsbug preferences (it does not delete the preference file, but empties the contents):

```
defaults delete tech.moddable.xsbug
```