Moddable-OpenSource/moddable #853: xsbug does not refresh the code in Windows.

js4iot, 2022-02-19:
**Build environment:** Windows
**Target device:** Moddable Two

**Description**
When I run `mcrun -d -m -p esp32/moddable_two` the *xsbug* window opens if it was closed. But if it was opened, the code that was compiled does not get automatically refreshed. The new code is executed in the board as expected, but the code displayed in *xsbug* is an older version. This means that tracing the code is weird because the lines do not match.

I have to close the file and re-open it. But this is usually after I have spent a few minutes trying to understand why my code is not behaving the way that I am expecting it to work :-)

**Steps to Reproduce**
1.  Write some sample code
2. Close *xsbug*
3. Build and the module using `mcrun -d -m -p esp32/moddable_two`
4. Don't close *xsbug*
5. Go back to your code and put a `debugger;` statement
6. Build and the module using `mcrun -d -m -p esp32/moddable_two`
7. The code stops as expected, but the line highlightedd is not `debugger;` (as expected) because the code did not refreshed.
8. Inside *xsbug* close the file
9. Re-open the file, it now shows the `debugger;` line. 

**Expected behavior**
Ideally, I do not need to close and open the file, this should happen automatically.

**Images**
[image]

wilberforce, 2022-02-19:
As a work around of you run Mcconfig again it should reload

andycarle, 2022-02-19:
> As a work around of you run Mcconfig again it should reload

@wilberforce Have you seen this issue also? I'm not able to reproduce it on my machine... Trying to guess what the trick may be.

js4iot, 2022-02-20:
I am using a Virtual Machine where I just install few tools related to Moddable SDK, including ESP-IDF, Moddable SDK, Visual Studio and VS Code. Nothing else is installed in this box. It’s very strange.

not sure if this part of the issue, or this is the expected behaviour. What happens when multiple files are opened? For example, while debugging it goes to some of my modules or some system modules. I would expect some tabs, one tab per document. But instead, I have to close this document for the other ones to show up.

phoddie, 2022-02-20:
Perhaps the Virtual Machine is getting in the way of xsbug receiving notifications that the source file has been editing. I'm reasonably certain that @andycarle isn't using a Virtual Machine. For future reference, which one are you using.

> not sure if this part of the issue, or this is the expected behaviour. What happens when multiple files are opened? For example, while debugging it goes to some of my modules or some system modules. [...]

It is not part of the issue. xsbug has one tab for each connected virtual machine. This is explained in the [xsbug documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/xsbug.md#machine-tabs)  and the [video demonstration of xsbug](https://youtu.be/vqu8gDV7AOo) the documentation links to.

phoddie, 2022-02-24:
@js4iot -- could you let us know what virtualization environment you are using? It seems like it could be relevant here.

Also, if possible, it would be really helpful if you could try to reproduce the problem on Windows directly, without any virtualization. That would help focus investigation into this report.

Thanks!

js4iot, 2022-02-25:
I am using [VirtualBox](https://www.virtualbox.org/), and I just realize that may be the actual problem. I just realized this morning that VS Code does not update the files to be committed to my repo either. I have to click on the refresh button. Although I am using a virtual machine, the files are located locally in the host computer, which means that they are accessed through a network drive. And that may not trigger the event to notify the apps (xsbug and VS Code) to refresh the files.

js4iot, 2022-02-25:
Even in the virtual machine... Xsbug refreshes automatically when I save a file in the C drive, but not in the network drive and I think this is expected.

Please feel free to close this issue, I will investigate with VirtualBox to see if there is a way to get notified when the files are saved in the network.

phoddie, 2022-02-25:
OK, great. That's very helpful to know. If you do find a solution, please post it here as it may help others. Thank you!

closed 2022-02-25 by phoddie