Moddable-OpenSource/moddable #854: SDK 4.4 esp32 warnings

wilberforce, 2022-02-21:
```
C:/Users/rhys/Projects/moddable/build/tmp/esp32/m5stick_c/debug/esp_host/xsProj-esp32/sdkconfig.mc:230 CONFIG_ESP_CONSOLE_UART_NONE was replaced with CONFIG_ESP_CONSOLE_NONE
C:/Users/rhys/Projects/moddable/build/tmp/esp32/m5stick_c/debug/esp_host/xsProj-esp32/sdkconfig.mc:297 CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE was replaced with CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE
C:/Users/rhys/Projects/moddable/build/tmp/esp32/m5stick_c/debug/esp_host/xsProj-esp32/sdkconfig.mc:298 CONFIG_ESP32_PHY_INIT_DATA_IN_PARTITION was replaced with CONFIG_ESP_PHY_INIT_DATA_IN_PARTITION
C:/Users/rhys/Projects/moddable/build/tmp/esp32/m5stick_c/debug/esp_host/xsProj-esp32/sdkconfig.mc:299 CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER was replaced with CONFIG_ESP_PHY_MAX_WIFI_TX_POWER
C:/Users/rhys/Projects/moddable/build/tmp/esp32/m5stick_c/debug/esp_host/xsProj-esp32/sdkconfig.mc:300 CONFIG_ESP32_PHY_MAX_TX_POWER was replaced with CONFIG_ESP_PHY_MAX_TX_POWER
```

Looks like these need to be updated in the defaults....

https://github.com/Moddable-OpenSource/moddable/blob/960c30d9e13027d50728cf65991bc996878db73a/build/devices/esp32/xsProj-esp32/sdkconfig.defaults#L230

phoddie, 2022-02-22:
Thanks for the report. We'll take care of that. Note that the ESP32, ESP32-S2 and ESP32-S3 all need to be updated.

wilberforce, 2022-02-24:
Confirmed working on a std esp32.

closed 2022-02-24 by phoddie