Moddable-OpenSource/moddable #857: Does the Instrumentation API in esp32 use different indexes?

fernandotetu, 2022-02-24:
I ran the instrumentation example `examples/base/instrumentation/xsuse` and got negative values. Then looking for it, I found on issue #435 that for esp32, the instrumentation indexes need to increase by 3.
Thus, I changed the example by replacing the Slot heap and Chunk heap sizes by 14 and 15, respectively.  For the most of returns in the algorithm, I received a positive value or zero, but for the first one, `Boolean` returns a negative value:

	log
	Boolean: -142 slots = -2272 bytes

Therefore, I would ask you about three situations: 
1. Are negative values ​​valid?
2. Is There documentation that describe the index increments?
3. In some cases like `Number`, `Object {}:`, `String from ROM` always return zero bytes, is that right?

*Environment
ESP-IDF v4.4 
device: Chip is ESP32-D0WDQ6 (revision 1) - esp32/nodemcu

phoddie, 2022-02-24:
Runtime instrumentation is implemented as an array of 32-bit integer values. The meaning of the index depends on the host to allow instrumentation provided to adapt to the specific microcontroller,  RTOS, and build options. For [example](https://github.com/Moddable-OpenSource/moddable/blob/e8efd64e40bd7d1d0edd743fb88243d4157a97d1/xs/platforms/esp/xsHost.c#L93-L100), an ESP32 has two CPU cores and an ESP32-S2 has only one. The information about each index (name, offset, units) is transmitted to xsbug to allow it to display the instrumentation graphs. This information is not currently available through an API at runtime.

There are two ways to determine the array indices.

1. Inspect the Moddable SDK source code. The instrumentation arrays for an ESP32 host are in [xsHost.c](https://github.com/Moddable-OpenSource/moddable/blob/e8efd64e40bd7d1d0edd743fb88243d4157a97d1/xs/platforms/esp/xsHost.c#L79-L123). That is concatenated with the instrumentation for XS  in [xsDebug.c](https://github.com/Moddable-OpenSource/moddable/blob/e8efd64e40bd7d1d0edd743fb88243d4157a97d1/xs/sources/xsDebug.c#L2252-L2279).
2. View the instrumentation information sent to xsbug. To do this, set mxTrace to 1 in [serial2xsbug.h](https://github.com/Moddable-OpenSource/moddable/blob/e8efd64e40bd7d1d0edd743fb88243d4157a97d1/tools/serial2xsbug/serial2xsbug.h#L91) and rebuild Moddable SDK tools. Then when you debug on a device over serial you will see the xsbug communication, including these lines immediately after the connection is established:

```
<xsbug>
  <instruments>
    <instrument name="Pixels drawn" value=" pixels"/>
    <instrument name="Frames drawn" value=" frames"/>
    <instrument name="Network bytes read" value=" bytes"/>
    <instrument name="Network bytes written" value=" bytes"/>
    <instrument name="Network sockets" value=" sockets"/>
    <instrument name="Timers" value=" timers"/>
    <instrument name="Files" value=" files"/>
    <instrument name="Poco display list used" value=" bytes"/>
    <instrument name="Piu command List used" value=" bytes"/>
    <instrument name="System bytes free" value=" bytes"/>
    <instrument name="Chunk used" value=" / "/>
    <instrument name="Chunk available" value=" bytes"/>
    <instrument name="Slot used" value=" / "/>
    <instrument name="Slot available" value=" bytes"/>
    <instrument name="Stack used" value=" / "/>
    <instrument name="Stack available" value=" bytes"/>
    <instrument name="Garbage collections" value=" times"/>
    <instrument name="Keys used" value=" keys"/>
    <instrument name="Modules loaded" value=" modules"/>
    <instrument name="Parser used" value=" bytes"/>
    <instrument name="Floating Point" value=" operations"/>
  </instruments>
</xsbug>
```

The <instruments> element contains <instrument> elements in order of the instrumentation indices. The data above is shown for an ESP8266 build.

fernandotetu, 2022-02-28:
Thanks Peter, it was very thoughtful of you to take the time to write a valuable and complete explanation bout our comments. 

How about negative values, have you any idea about them? Also, could you explain why some information does not take up space (0 bytes) in the memory ram?

phoddie, 2022-02-28:
> How about negative values, have you any idea about them?

Please provide an example that shows how you are obtaining negative values. From the example above, it is not clear.

> In some cases like Number, Object {}:, String from ROM always return zero bytes, is that right?

It is a bit complicated, but:

- A JavaScript Number value is always stored in a slot. A number in a local variable, therefore, only takes space on the stack not in any heap. A Number assigned to a property of an object, however, requires an entry in the slot heap.
- A JavaScript String value stored in ROM uses no RAM. As with a Number, if it is assigned to a local variable on the stack, no heap memory is used, but assigning it to the property of an object uses a slot.
- A JavaScript Object stored in ROM uses no RAM. A JavaScript Object allocated at runtime, uses slot memory.

fernandotetu, 2022-03-01:
Please provide an example that shows how you are obtaining negative values. 

I only performed the xsuse example from the instrumentation module. I had to modify the lines 89-90 and 96-97 in order to adjust the index to work on esp32. As a result, the first line related to the boolean entry returns a negative value. 

Moreover, I performed a new installation of Moddable SDK and ESP-IDF v4.4 in a different environment (Linux Fedora 29 - the first report had happened on macOS). Finally, I conducted the same process to adapt the xsuse example described above that results, once again, in a negative value for boolean entry. Nonetheless, the negative value was different; take a look:

	logs
	Linux: Boolean: -141 slots = -2256 bytes
	MacOS: Boolean: -144 slots = -2304 bytes

The others entries on xuse return the same value on both platforms.  

Hardware 
NodeMCU ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None

phoddie, 2022-03-02:
The [xsuse](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/base/instrumentation/xsuse/main.js) example has been updated:

- Attempts to determine the offset of the XS instrumentation at runtime to eliminate the dependency on the device
- Determines the slot size at runtime
- Fixes measurement of an empty object
- Performs an initial GC to eliminate negative values on first run because of heap growth

In addition mcsim has been updated so that it works with `Instrumentation.get` and consequently `xsuse` may be run on the desktop. To run with mcsim (after updating and rebuilding Moddable SDK tools):

```
mcconfig -d -m -p sim
```