Moddable-OpenSource/moddable #861: Is it possible to change/get the esp8266's CPU frequency?

fernandotetu, 2022-03-01:
Hi there,

The esp8266 should be run at the default of 80MHz CPU frequency but it can be change until 160MHz. I need ensure what is the current CPU frequency for my experiments. 
So, I would ask you about it, is it possible get/set the current CPU frequency through Moddable SDK?  

I have been using  ESP8266 NodeMCU module.

phoddie, 2022-03-01:
Yes, using the XS in C API you can access all of the native functionality of the ESP8266. That includes changing the CPU frequency. The [XS in C](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md) documentation provides a detailed reference for how to bind native code to the XS JavaScript  engine. The final chapter in our book [IoT Development for ESP32 and ESP8266 with JavaScript](https://www.moddable.com/book) is a good introduction to the techniques.

You'll use XS in C to create a function that calls this function in the ROM:

```
extern "C" void ets_update_cpu_frequency(int freqmhz);
ets_update_cpu_frequency(80);
```

phoddie, 2022-03-01:
A [recent exchange](https://gitter.im/embedded-javascript/moddable?at=62144844d1b64840db32c368) on our Gitter about binding in the native restart function is relevant here.

closed 2022-03-07 by phoddie

mauroForlimpopoli, 2023-04-18:
Good morning,
I created a file .C and I add this function
```
void xs_putSlow(xsMachine * the)
{
	ets_update_cpu_frequency(80);
	int a = 2;
	xsResult = xsInteger(a);
}
```
in the main after
```
function putSlow() @ "xs_putSlow";   ////cpu 80 mhz ??
trace (putSlow())
```
I get 2, , so it works correctly, but the power consumption (with WIFI on) is always about 130mA.
Where is the error?
Is possible otherwise to start the entire project with clock to 80 mHZ or less

Thank you

phoddie, 2023-04-18:
Hello. My understanding is that the default CPU frequency on the ESP8266 is 80 MHz. If you want to run at 80 MHz, you don't need to call anything.

mauroForlimpopoli, 2023-04-19:
It's for ESP32.

phoddie, 2023-04-19:
Thanks for the clarification. I think this might not be the best repository for this topic. You are already able to call the ESP-IDF native functions from your JavaScript. The question now, if I understand well, is how to use the ESP-IDF to achieve your energy management goals.

That's a complicated topic. From experience, I know that just changing the clock speed isn't likely to be enough -- and change the clock speed can break other subsystems which rely on the original clock speed. I suggest you consult the ESP-IDF documentation and examples as a starting point. If that's not enough, the ESP-IDF forum might be a good place to post your question.

I'm happy to, of course, help with Moddable SDK integration questions related to this topic.

salarizadi, 2023-04-19:
> It's for ESP32.

Hello.
It is better to do this in the sdkconfig.h file. For more guidance on changing the sdkconfig, see the following document

SDK configuration:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html

CPU:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html

mauroForlimpopoli, 2023-04-21:
In the file sdkconfig,default I changed 
```
CONFIG_ESP32_DEFAULT_CPU_FREQ_80=n
CONFIG_ESP32_DEFAULT_CPU_FREQ_160=n
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240
```
in 
```
CONFIG_ESP32_DEFAULT_CPU_FREQ_80=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_160=n
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=n
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=80
```
It seems to works but I can't verify it
Surely I expected more current consumption reduction. I got only 10/20 mA 
With wifi active I gone from 140 to 120 mA (not stable)
This is my test
regards