Moddable-OpenSource/moddable #869: Typescript module build failed. wrong module duplicate path

MKGaru, 2022-03-11:
**Build environment:**  Linux
**Target device:** NodeMCU ESP32,

**Description**

mkdir SIMPLE  (or `xs-dev init SIMPLE`)
main.js
```javascript
import storage from "storage"
export default async function () {
  storage
}
```
storage.js
```javascript
class Storage {
}
Object.freeze(Storage)
export default new Storage()
```
manifest.json
```
{
  "include": [
    "$(MODDABLE)/examples/manifest_base.json"
  ],
  "modules": [
    "*": [
      "./storage",
      "./main"
    ]
  ]
}
```

It build successed.   (`mcconfig -d -m -p esp32/nodemcu` or `xs-dev run --device=esp32/nodemcu`)
But change to typescript.

```bash
mv main.js main.ts
mv storage.js storage.ts
```
build wrong path in makefile.

```
$(MODULE_DIR)/storagexsb: $(MODULES_DIR)//SIMPLE/storage.js
  @echo "#xsc storage.xsb"
  $(XSC) $(MODULES_DIR)//SIMPLE/storage.js -d -c -e-o $(@D) -r storage
$(MODULE_DIR)/main.xsb: $(MODULES_DIR)//SIMPLE/main.js
  @echo "#xsc main.xsb"
  $(XSC) $(MODULES_DIR)//SIMPLE/main.js -d -c -e-o $(@D) -r main
```
[image]

phoddie, 2022-03-11:
Thank you for the excellent report and helpful debugging notes! We'll take a look.

phoddie, 2022-03-14:
This should be resolved now. Please confirm. Thank you!

MKGaru, 2022-03-15:
Awesome! It's works. Thank you.

closed 2022-03-24 by phoddie