Moddable-OpenSource/moddable #870: WiFiModule Allow PSK instead of password

MKGaru, 2022-03-13:
**Describe**
WiFi Module Allow PSK instead of password

**Why do you think this feature would be useful?**
There is a slight security advantage as it avoids storing raw passwords.

**Proposal change**

https://github.com/Moddable-OpenSource/moddable/blob/5df3b556aa18cdefc03fd076b735ac26b4178aeb/modules/network/wifi/esp/modwifi.c#L133

https://github.com/Moddable-OpenSource/moddable/blob/5df3b556aa18cdefc03fd076b735ac26b4178aeb/modules/network/wifi/esp32/xs6wifi.c#L183

remove -1  both files.
(both source use memcpy, not strcpy , because maybe not required terminate null charactor)

I tried on esp32, it works.

ref.
http://jorisvr.nl/wpapsk.html
esp8266/Arduino/pull/1850

phoddie, 2022-03-15:
Interesting. It is nice that the password wouldn't need to be stored, though the PSK value still allows log-in to the network, so that should still be secured in a real product.

It seems like the password and PSK are not quite the same. I'm not sure if there should be a different name in JavaScript to distinguish those (`config.psk` instead of `config.password`). If everyone else is already overloading password, maybe we should do the same.

wilberforce, 2022-03-15:
You do you generate the PSK?

wilberforce, 2022-03-15:
https://www.wireshark.org/tools/wpa-psk.html
[pbkdf2.js](http://anandam.name/pbkdf2/) might be useful if - then the same ssid and password in the manifest could be used and they could be converted to the psk for use on the device.

phoddie, 2022-10-20:
Sorry to be so slow The proposed change makes sense. Even macOS allows a 64 character password, so we should just always do that. This will be included in tomorrow's Moddable SDK update. Thank you!

phoddie, 2022-10-21:
Closing as the proposed change has been [applied](https://github.com/Moddable-OpenSource/moddable/commit/731df5c8f807240c28cb84413b8f9e5eb13a393d).

closed 2022-10-21 by phoddie

MKGaru, 2022-10-21:
> 提案された変更が[適用さ](https://github.com/Moddable-OpenSource/moddable/commit/731df5c8f807240c28cb84413b8f9e5eb13a393d)れたため、終了します。

Thank you so much!
By the way, you applied for access point ssid & password, but i think should be change sta mode too.

phoddie, 2022-10-22:
😱

phoddie, 2022-10-24:
I tried again. ;) Any better?

MKGaru, 2022-10-25:
Excellent!
It's works.
Thank you for update.