Moddable-OpenSource/moddable #871: Setup modules not loaded when running in a simulator

FWeinb, 2022-03-13:
**Build environment:** macOS
**Target device:** Simulator/m5paper

**Description**
I want to add support for the [`global.power`](https://github.com/Moddable-OpenSource/moddable/blob/5df3b556aa18cdefc03fd076b735ac26b4178aeb/build/devices/esp32/targets/m5paper/setup-target.js#L6-L19) object to the simulator but as the [`Control`-class relies on `screen`](https://github.com/Moddable-OpenSource/moddable/blob/5df3b556aa18cdefc03fd076b735ac26b4178aeb/build/simulators/modules/control.js#L14) I need to execute this during setup. But when adding a `setup/target` module to the `preload`-object in the [manifest.json of the m5paper simulator](https://github.com/Moddable-OpenSource/moddable/blob/public/build/simulators/m5paper/manifest.json) like this:
```js
{
  "modules": {
    	// ...
		"setup/target": "./setup-target"
  },
  "preload": [
 	//  ...
    "setup/target"
  ],
  // ...
}
```
with `setup-target.js`
```js
export default function () {
  trace(`Setup-Target called\n`);
  done();
}
```
the default exported function is never called.  

**Steps to Reproduce**
1. Copy the content of [these files](https://gist.github.com/FWeinb/a83e2a30e26cea0c01ceb4902f3bb35a) to `$MODDABLE/build/simulators/m5paper`
2. Run any of the `epaper-*` examples using `mcconfig -d -m -p sim/m5paper`  
3. See that `Setup-Target called` is never printed in the xsbug output.

phoddie, 2022-03-15:
Setup modules were created to set-up the runtime environment ahead of execution of `main`. They take care of things like initializing the display and connecting to the network. Those things are are done by the host operating system when running on a computer. The documentation alludes to this: "The simulator implicitly performs the work of setup already..".

It should be fine to support `setup/*` modules the simulator too. That can be useful in some scenarios, and having a consistent mechanism makes sense. Your changes above to do that look very reasonable. If you would bundle that into a PR, we can review and merge it. Thanks!

phoddie, 2022-03-24:
This has been resolved with the PR from @FWeinb (thank you!).

closed 2022-03-24 by phoddie