Moddable-OpenSource/moddable #875: Timer: Timer callbacks are getting mistriggered repeatedly

Arorar3, 2022-03-16:
**Environment**

* Development Board: ESP32
* ESP-IDF version: v4.4
* Moddable SDK version: d085d3a6
* Build System: mcconfig
* Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
* Operating System: Mac
* Using an IDE?: No
* Power Supply: 12V DC
* Router: Mobile Hotspot

**Problem Description**
  
   1. The timer module behaves incorrectly after overflowing variables. The timer behaves incorrectly in two scenarios: 
        1) when the trigger time gets overflow and
        2) when the internal tick variable overflows.  
   2. Timer module behaves as follows when a variable overflows:
       1) the callback gets repeated.
       2) the callback stays uncalled for a while.
   3. Due to repeated callbacks, moddable exhibits the following issues:
      1) Multiple timer callbacks are getting call repeatedly within 1 second.
      2) application gets stuck for a while. 

**Steps to reproduce**

Here are the steps to reproduce the issue in [MODDABLE](https://github.com/Moddable-OpenSource/moddable)

1. In [xsHost.h](https://github.com/Moddable-OpenSource/moddable/blob/public/xs/platforms/esp/xsHost.c), we must declare the OFFSET micro.
[image]

2. In [xsHost.c](https://github.com/Moddable-OpenSource/moddable/blob/public/xs/platforms/esp/xsHost.c), we need to add OFFSET with the current modMilliseconds clock. So that timer module will overflow after 40 secs.
[image]

3. I took mqttbasic example to produce this timer overflow behavior. In the example, a timer is getting call every 1 second and publishing mqtt request.
[image]

**Debug Logs**

1. Continuous callbacks - after 40 secs of running, 31 continuous calls to timer callback functions within 1 second of duration.

[image]

2. Application getting stuck - After 31 continuos calls  within 1 second, application gets stuck for around 31 seconds and then resumes.

[image]

**Analysis:**

1. In [xsHost.c](https://github.com/Moddable-OpenSource/moddable/blob/public/xs/platforms/esp/xsHost.c), **modMilliseconds** API provides the tick to the timer module, which is an **unsigned integer** (0, 4294967295) thus overflows after 49 days, 17 hours and 2 mins. 

[image]

However, in the **modTimersExecute** function of [Timer.c](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/base/timer/mc/timer.c) this **modMilliseconds** function gets assigned to **now** variable which is an **signed integer** (-2147483647, +2147483647). 
The timer structure modTimerRecord has a variable **triggerTime**  that also is of type **signed integer**. Both of these will both overflow after 24 days, 20 hours and 31 mins.

[image]

 Furthermore, when the timer callback is called at every specified interval, the **triggerTime** is updated and it overflows when it is near to the maximum value of **signed integer** (24 days, 20 hours and 31 mins).  

[image]

2. In the mqttbasic example, timer gets called every 1 second(1000 millisecs) interval. modMilliseconds() provides the current time as time ticks from boot up + offset which is close to **signed integer** overflow limit. After 40 seconds of running when timer executes, it sets the triggertime variable to next interval. At this point triggerTime variable crosses its limit from +2147483647 and goes in negative value. 
After that whenever the **modTimersExecute** function is called, **(walker->triggerTime <= now)** becomes true and the callbacks are repeated.

[image]

**Proposed solutions:**
1. The first solution is to put a condition in **modTimersExecute** function while iterating the global timers. The condition is to check the timers when **walker->triggerTime** overflows its range and goes in negatives and **now** does not. And when that situation happens then continue the loop and move to next timer.
[image]

2. The second solution is increase the timer variables from integer(32 bits) type to long integer(64 bits) type and postpone this problem to happen to thousand of years.
Code changes are below(attached):

2.1. Change the **modMilliseconds()** to provide current ticks from 32 bits to 64 bits.
[image]

2.2. Update the **modMilliseconds()** in xsHost.h file.

[image]

2.3. Change the modTimerRecord structure. Update **triggerTime** variable  from 32 bits integer to 64 bits signed integer. This will increase the structure size to 40  bytes, so we are using #pragma pack(2) to pack structure members with 2 bytes alignment.
[image]

2.4. Update the **now** variable in **modTimerExecute()** to 64 bits signed integer type.

[image]

2.5. Update the modTimerNext(). Update the **now** and **delta** variables to 64 bits signed integers.
[image]

**Debug Logs**
1. [Failure log (moddable)](https://github.com/Moddable-OpenSource/moddable/files/8271950/teraterm_Pub_Moddable.1.log)

**attachment**
[Solution2_CodeDiff.zip](https://github.com/Moddable-OpenSource/moddable/files/8273617/Solution2_CodeDiff.zip)

phoddie, 2022-03-17:
Thank you for the report, @Arorar3!

The second proposed solution has some issues:

- Increases memory used by each timer
- Increases execution time by promoting 32-bit math to 64-bit math in calculations
- Isn't portable to all targets that use this timer implementation (includes changes only for ESP32)

The first proposed solution has none of these issues and is considerably simpler. If the first proposed solution works in all cases, it isn't clear why the alternative is offered. Is there a known issue with the first proposed solution that the second proposed solution solves?

Arorar3, 2022-03-23:
Thanks for the response!

The issue happens when **triggerTime** overflows its integer range and goes in negatives and after that whenever the modTimersExecute function is called, condition **(walker->triggerTime <= now)** becomes true and the callbacks are unnecessarily repeated.

With the first proposed solution, it will handle this repeated callbacks scenario but what we have observed that when this overflows happens there are frequent (in 0-2 milliseconds interval) calls to **modTimerExecute()** function for some iterations. That would affect the performance.

To avoid that we have come up with second solution in which we have increased timer variable size from 32 bits to 64 bits so the overflow will occur after thousand of years.
Answering to the concerns you have regarding the second proposed solution.
- Yes, it will increase memory. With the size change in **triggerTime** variable in **modTimerRecord** structure, the structure size was becoming 40 bytes so we have used #pragma pack(2) to pack structure members with 2 bytes alignment. With this each timer will take 2 bytes extra. 
- Yes, it will require some more execution time.
- Yes, it may not be portable to all the targets but with this solution we are bypassing the variables overflow issue and the drawback of first proposed solution where we have observed frequent calls to **modTimerExecute()**.

phoddie, 2022-03-23:
Thank you for your notes.

> ...the structure size was becoming 40 bytes so we have used #pragma pack(2) to pack structure members with 2 bytes alignment.

This seems likely to break targets that have a 4-byte alignment requirement , including the Xtensa in ESP8266 and lower-performance ARM cores.

It does not appear that there is a viable solution here. The first solution causes unnecessary CPU load; the second has multiple issues, most notably that it is not portable.

closed 2022-05-02 by phoddie