Moddable-OpenSource/moddable #876: function `name` is set incorrectly on a method for a property key that is a registered symbol

gibson042, 2022-03-16:
**Environment:** XS 11.6.0 32 4

**Description**
XS incorrectly set the `name` of a function created as a method for a property key that is a registered symbol.

**Steps to Reproduce**
```js
const uSym = Symbol("unique symbol");
const rSym = Symbol.for("registered symbol");
const obj = { [uSym](){}, [rSym](){} };
const uFn = obj[uSym];
const rFn = obj[rSym];
[uFn, rFn].map(fn => `${fn} // => ${JSON.stringify(fn.name)}`).join("\n");
// =>
// function ["[unique symbol]"] (){[native code]} // => "[unique symbol]"
// function ["registered symbol"] (){[native code]} // => "registered symbol"
```

**Expected behavior**
Both functions should have a `name` value with enclosing brackets, per step 2.c of [SetFunctionName](https://tc39.es/ecma262/#sec-setfunctionname) (invoked from step 2 of [MethodDefinitionEvaluation](https://tc39.es/ecma262/#sec-runtime-semantics-methoddefinitionevaluation) with a Symbol-valued name from [DefineMethod](https://tc39.es/ecma262/#sec-runtime-semantics-definemethod)).

**Actual behavior**
Only the non-registered Symbol description is wrapped in brackets.

gibson042, 2022-03-17:
@dckc I just opened https://github.com/tc39/test262/issues/3436 , and anyone feeling ambitious can submit a PR to fix it (although that likely won't be me, at least in the near future).

phoddie, 2022-03-17:
@gibson042, thank you for another excellent report. The test case reproduces the behavior here too. I'm not certain yet when we'll get to look into a fix.

closed 2022-05-02 by phoddie