Moddable-OpenSource/moddable #878: File rename broken on SPIFFS with a filename that has a "/"

cmidgley, 2022-03-17:
**Build environment:** Linux
**Target device:** ESP32/heltec_lora_32

**Description**
When using ESP32 SPIFFS, and renaming to a filename with a slash in the name (which should just be part of the filename, and not represent any real path in the flat SPIFFS file system), the rename code attempts to manage the path resulting in an error.

**Steps to Reproduce**
```js
import {File, Iterator, System} from "file";
import config from "mc/config";

const root = config.file.root;

function ren(from, to) {
    let file = new File(root + from, true);
    file.close();

    File.rename(root + from, root + to);

    if (File.exists(root + to))
        trace(`${from} renamed to ${to}\n`);
    else
        trace(`Rename failed: ${from} to ${to}\n`);
}

// rename a regular file - this works
ren("test.txt", "test2.txt");

// rename the fake directory file - the rename succeeds, but the file isn't found (the leaf name is stripped off the name)
ren("fakedir/test.txt", "fakedir/test2.txt");
```

**Expected behavior**
The resulting filename should retain full name, including the slashes.

**Other information**
In the ESP32 implementation, the bug is in [this code](https://github.com/Moddable-OpenSource/moddable/blob/8bd7a2fdc470ff1c0ceb500f801760fe463400bd/modules/files/file/esp32/modFile.c#L259-L265).  

Modified to the following fixes the issue (though it could be optimized and avoid the `xsmcToStringBuffer` on `toPath` for the SPIFFS use case):

```js
void xs_file_rename(xsMachine *the)
{
    char* path;
    char toPath[MAX_FILENAME_LENGTH + 1];
    int32_t result;
    char* slash;
    size_t pathLength = 0;

    path = xsmcToString(xsArg(0));
#if MODDEF_FILE_FAT32
    slash = c_strrchr(path, '/');
    if (slash){
        pathLength = slash - path + 1;
		if (pathLength >= sizeof(toPath)) xsUnknownError("path is too long");
        c_memcpy(toPath, path, pathLength);
        toPath[pathLength] = '\0';
    }
    xsmcToStringBuffer(xsArg(1), toPath + pathLength, sizeof(toPath) - pathLength);
    path = xsmcToString(xsArg(0));
#else
    xsmcToStringBuffer(xsArg(1), toPath, sizeof(toPath));
#endif

    startFS();

    result = rename(path, toPath);

    stopFS();

    xsResult = xsBoolean(result == 0);
}
```

I did not check the other implementations of `File` to see if similar bugs exist.

phoddie, 2022-03-21:
Slashes and SPIFFS! There could be an issue. But first, the test code above doesn't quite match the [documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/files/files.md#static-renamefrom-to). I'm not sure if that is contributing to the problem. I'd like to sort that out before diving into the implementation.

`File.rename()` isn't the same as the `mv` terminal command. The second argument is only the file name, never a path (full or partial). The documentation shows:

```js
File.rename(config.file.root + "test.txt", "betterName.txt");
```

The example above applies `config.file.root` to both parts:

```js
File.rename(root + from, root + to);
```

cmidgley, 2022-03-21:
If your point is simply that the target of the rename should not have `config.file.root`, that's fine and makes sense given the rename-only characteristic of rename.  But the munging of slashes in rename on SPIFFS makes no sense, as SPIFFS is flat and slash is just a character in a filename like any other.  Rename is the only call that is challenged with slashes on SPIFFS (I now have unit testing running against all of them), and I think - at least conceptually - that the `#if` calls are needed like used elsewhere in the code to deal with the flat file system.

phoddie, 2022-03-21:
I'm not sure what the correct behavior is yet. I'm trying to understand.

Completely agree: a slash on SPIFFS file name is just a character. The first argument to rename is a path: the `root` prefix adds a slash at the start.

On the ESP32, the implementation uses `rename` which takes a path for both arguments. For a FAT32 file system, `xs_file_rename` could reject file names with a slash (in the second argument). But that's impractical because it doesn't know that it is a FAT32 file system or SPIFFS or something else (any code can mount a file system and it will be used). So, that's an inconsistency though to be lived with.

But that's also why we can't have a conditional on `MODDEF_FILE_FAT32` in rename. 

The code you propose to disable was originally written for ESP8266 which only supports SPIFFS. It puts the root prefix on the file name (second argument) so that SPIFFS  receives full paths for both arguments to `SPIFFS_rename` It does that by looking for the last slash, assuming that is the end of the prefix. That assumption is incorrect if the files name has a slash.

For ESP32, your change could be fine in all cases – the file system will sort it out. If ESP8266 applied your change, it will break the call to `SPIFFS_rename`. There we might get away with taking the first slash instead of the last slash. That means the prefix on ESP8266 is limited to one path segment, but that seems fine. Desktop platforms behave mostly like ESP32, so no changes there either. Since desktop file systems always support directories, they could safely throw when the new name argument contains a slash.

cmidgley, 2022-03-21:
Interesting problem w/ESP8266 - though honestly not sure I followed it.  I guess the question is, what's the solution?  Conditional on ESP32?  Some other answer?  Would prefer, if reasonable, to have my usage of `File` not require a patch to Moddable to use it.

If this turns into a mess, then I can find alternatives (not using slash as my separator for example), though that adds a bunch more code on my side to translate these paths, as my system adapts `File` to be emulating directories, and works on file systems outside of Moddable (NodeJS specifically).

phoddie, 2022-03-21:
Thanks for following along. I'm not entirely sure I follow either. ;) 

We both prefer a solution that works correctly everywhere, and doesn't require a patch. If we make the simplifying assumption that the root on ESP8266 is a single segment (one slash) then everything seems straightforward. I need to find some time to test that though.

phoddie, 2022-03-22:
After running some test cases on ESP8266 and ESP32, things are more in focus.

The proposed patch only works because the second argument to `rename` is a full path. The patch breaks the existing behavior of `File.rename()` because the ESP-IDF `rename` function does not accept a filename for the second argument: it requires a full path.

The challenge here:

- Continue to accept the defined behavior of `File.rename`: the first argument is a full file path and the second argument is a file name
- Allow rename on the flat-file system SPIFFS when there are slashes in the file name
- Have `File` behave predictably across devices and file systems

Consider this case:

```js
const root = config.file.root;
File.rename(root + "foo/a.b", "b.a")
```

The expected results is for `foo/a.b` to be renamed `foo/b.a`. That is what happens today's both on SPIFFS and FAT32. That should not change. (It is true that on SPIFFS there are no real directories so this is somewhat artificial; the existing behavior seems more helpful than harmful.)

This fragment should fail, because the destination is a partial path not a file name. 

```js
const root = config.file.root;
File.rename(root + "foo/a.b", "bar/b.a")
```

Today there is no validation on the second argument, so the behavior is host dependent.

I suggest we extend `File.rename()` so that when the second argument is a full path, it is passed through to the host file system. That allows this to work both for SPIFFS and FAT32.

```js
const root = config.file.root;
File.rename(root + "foo/a.b", root + "bar/b.a")
```

That's a small expansion of the definition of `File.rename()` to allows access to more of the capabilities of the host file system without breaking the current behavior.

Here' a modified version of `xs_file_rename` for ESP32 that implements this behavior. There is a corresponding change on ESP8266.

```c
void xs_file_rename(xsMachine *the)
{
    char* path;
    char toPath[MAX_FILENAME_LENGTH + 1];
    int32_t result;

	xsmcToStringBuffer(xsArg(1), toPath, sizeof(toPath));
	path = xsmcToString(xsArg(0));
	if ('/' != toPath[0]) {
		if (c_strchr(toPath + 1, '/'))
			xsUnknownError("invalid to");

		char *slash = c_strrchr(path, '/');
		if (!slash)
			xsUnknownError("invalid from");

		size_t pathLength = slash - path + 1;
		if (pathLength >= (c_strlen(path) + sizeof(toPath)))
			xsUnknownError("path too long");

		c_strcpy(toPath, path);
		xsmcToStringBuffer(xsArg(1), toPath + pathLength, sizeof(toPath) - pathLength);
	}

    startFS();

    result = rename(path, toPath);

    stopFS();

    xsResult = xsBoolean(result == 0);
}
```

phoddie, 2022-03-22:
> IMHO, with SPIFFS, the slash character should be no different than any other character. The file system is flat, and should be treated that way. 

If `File` was a SPIFFS-only API, that would be reasonable. But, it isn't. On ESP32, there can be several volumes, each with a different file system. The behaviors across those volumes should be predictable without the calling code having to know the file system.

cmidgley, 2022-03-22:
I was unaware File could support multiple volumes.  I just scanned the doc again and I can't find a mention of volumes (the word 'volume' isn't used), and I see what appears to be a modal switch from SPIFFS to FAT at the end of the page, but didn't discover how it supports multiple file systems.  

What is the syntax to access multiple file system on the same host at the same time?  Is it a different prefix like `config.file.root` but pointing to a different FS?  

Or do you mean that different compiled versions of the code, depending on the features selected in `manifest.json`, might use a different file system?  In that case, I don't yet see why my proposal doesn't work.

I think it is odd to require `config.file.root` on the target of a rename when on SPIFFS, but not allowing it when on FAT.  I was trying to avoid that situation - so the same code works on all file systems.  The difference of course is that you should not expect directories on a flat file system, but that requirement permeates usage across all APIs, not just rename.

phoddie, 2022-03-22:
Support for volumes is provide through the path on hosts that support multiple volumes. On macOS, for example, the built-in drive is usually accessed as `/Volumes/Macintosh HD/`.

ESP-IDF has a virtual file system with file system plug-ins that provide a similar capability. They begin at `/`. The volume mounted by the Moddable SDK by memory mapping a partition is at `config.file.root` (which is `/mod/`). There can be other volumes, for example an SD Card, mounted by other means. They will be accessible at something like `/sdcard/`. The Moddable SDK `File` API works with them as well. Each volume can have a different file system. A flash partition might use SPIFFS while the SD Card uses FAT32.

> I think it is odd to require config.file.root on the target of a rename when on SPIFFS, but not allowing it when on FAT

It is. If I suggested that, it was my mistake. Apologies. The root is never required on the target, and the requirements are the same for FAT32 and SPIFFS.

The following two lines should give the same result. The proposal allows both to work on both file systems.

```js
File.rename(root + "foo/a.b", "b.a");
File.rename(root + "foo/a.b", root + "foo/b.a");
```

It also allows the situation on ESP32 where ESP-IDF `rename` is used to move directories on FAT32. This also gives the same result on SPIFFS.

```js
File.rename(root + "foo/a.b", root + "bar/b.a");
```

phoddie, 2022-03-23:
> in the last example, if root is not included on the target, will SPIFFS still rename correctly?

No. In fact, the implementation of `xs_file_rename` above explicitly rejects it:

```c
		if (c_strchr(toPath + 1, '/'))
			xsUnknownError("invalid to");
```

There is problem with a slash in the second argument if it not a full path. The merging of an absolute path with a partial path is relative to the directory of the absolute path.  Given `File.rename(root + "foo/a.b", "bar/b.a")`  the results is not `root + "bar/b.a` but `root + "foo/bar/b.a"`. Making the second argument a full path eliminates the need to merge the paths, and conveniently matches a valid way to call the native `rename` function in ESP-IDF (and other POSIX compliant systems).

> I don't quite see the importance of making root optional on target. 

The root is not allowed on target today. The target argument is defined as a filename, not an absolute or relative path. This may not always be rigorously enforced, but it is the defined behavior. Changing it would be incompatible. Adding the root is is a clear signal that the caller is opting in to a different behavior.

> Should I take the code proposed and run it through my unit tests?

Please. That would be great!

cmidgley, 2022-03-23:
It passes the rename unit test with flying colors.  The tests confirm that it can rename, and that renames work with and without slash characters (that they are treated as a standard character).    It does not test any assumptions about FAT or the target name being used for anything other than SPIFFS.

cmidgley, 2022-03-23:
Fantastic.  Thanks Peter!

phoddie, 2022-03-28:
This has been resolved.

closed 2022-03-28 by phoddie