Moddable-OpenSource/moddable #879: Change Date.parse to round milliseconds down

gibson042, 2022-03-17:
**Describe**
XS appears to be the only ECMAScript implementation with round-to-nearest millisecond `Date.parse` (and by extension, `new Date(value)`) behavior:
```sh
$ eshost -se '
  [
    "1970-01-01T00:00:00.000500001Z",
    "1969-12-31T23:59:59.999515625Z",
    "1969-12-31T23:59:59.999015625Z",
  ]
  .map(str => {
    const tv = Date.parse(str);
    return `${str} => ${(new Date(str)).toISOString()} (${Object.is(tv, -0) ? "-0" : tv} ms from epoch)`;
  })
  .join("\n")
'
#### ChakraCore, engine262, GraalJS, Hermes, SpiderMonkey, V8
1970-01-01T00:00:00.000500001Z => 1970-01-01T00:00:00.000Z (0 ms from epoch)
1969-12-31T23:59:59.999515625Z => 1969-12-31T23:59:59.999Z (-1 ms from epoch)
1969-12-31T23:59:59.999015625Z => 1969-12-31T23:59:59.999Z (-1 ms from epoch)

#### JavaScriptCore
1970-01-01T00:00:00.000500001Z => 1970-01-01T00:00:00.000Z (0.500001 ms from epoch)
1969-12-31T23:59:59.999515625Z => 1970-01-01T00:00:00.000Z (-0.484375 ms from epoch)
1969-12-31T23:59:59.999015625Z => 1970-01-01T00:00:00.000Z (-0.984375 ms from epoch)

#### Moddable XS
1970-01-01T00:00:00.000500001Z => 1970-01-01T00:00:00.001Z (1 ms from epoch)
1969-12-31T23:59:59.999515625Z => 1970-01-01T00:00:00.000Z (0 ms from epoch)
1969-12-31T23:59:59.999015625Z => 1969-12-31T23:59:59.999Z (-1 ms from epoch)
```
I think it would be better to round down (i.e., towards negative infinity), yielding results equivalent to ignoring all digits beyond millisecond precision.

**Why do you think this feature would be useful?**
It would align with the majority of other implementations, and completely eliminate a unique behavior from the landscape. And because rounding up date-time values is relatively uncommon, it would likely also reduce developer surprise (*especially* when rounding crossing coarse boundaries, such as the above example where parsing a "1969-…Z" string returns a time value in 1970).

**Describe alternatives you've considered**
The current behavior is technically compliant with the underspecified https://tc39.es/ecma262/multipage/numbers-and-dates.html#sec-date.parse , and there is no imperative to change it. I just think that doing so would be useful.

phoddie, 2022-03-17:
Thanks for the report. No question that it is desirable to have a consistent behavior across engines here. We'll have to take a look to see what's involved.

phoddie, 2022-03-17:
Maybe we just need to change from `c_round` to `c_trunc` [here](https://github.com/Moddable-OpenSource/moddable/blob/public/xs/sources/xsDate.c#L268). At least that gives the expected result.

gibson042, 2022-03-18:
Yes, that looks right to me. Or alternatively, increasing `aResult` only while `fraction >= 1` (provided `theCharacter` and `theString` still advance past all the digits).

closed 2022-05-02 by phoddie