Moddable-OpenSource/moddable #895: modulo signed zero incorrect?

devsnek, 2022-04-04:
Using the test binary release downloaded from your github repo:

```
$ eshost -ts -e "Object.is(-120 % 2, -0)"
┌────────────────┬───────┐
│ ChakraCore     │       │
├────────────────┼───────┤
│ engine262      │ true  │
│ JavaScriptCore │       │
│ QuickJS        │       │
│ SpiderMonkey   │       │
│ V8             │       │
├────────────────┼───────┤
│ XS             │ false │
└────────────────┴───────┘
```

I just noticed this while testing something else.

devsnek, 2022-04-04:
I think this is step 8 of https://tc39.es/ecma262/#sec-numeric-types-number-remainder? not 100% sure

phoddie, 2022-04-04:
Thanks for the report.

The integer optimizations in XS take priority, which causes this incompatibility with the spec. A quick fix would be to change

https://github.com/Moddable-OpenSource/moddable/blob/e26597b9bf71e4d79f01cb329bd04f1562d32ddc/xs/sources/xsRun.c#L3400-L3402

...to...

```c
			if ((slot->kind == XS_INTEGER_KIND) && (slot->value.integer >= 0)) {
```

That forces it through the floating point case for a negative dividend, which generates the correct result.

There is at least one other `-0` related issue open. We're still considering if we should make them always work, or have a build option to enable that for situations where it matters. It is a trade-off of performance & code size versus strict compatibility.

phoddie, 2022-06-06:
Fixed when `mxMinusZero` is defined for the build (and `xst` will set `mxMinusZero` moving forward to prioritize conformance over performance).