Moddable-OpenSource/moddable #897: Moddable documenation Windows update error

louisvangeldrop, 2022-04-08:
In the windows update documentation:
https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#win-update

The following error occurs: 

Updating
The Moddable SDK tools are frequently updated with improvements and added functionality. You should occasionally update your host environment by following these steps:

Update your local clone of the [Moddable repository](https://github.com/Moddable-OpenSource/moddable). For instance, using Git Bash:

cd $MODDABLE : **_Should be %MODDABLE%_**
git pull
Note that if you have any local changes to Moddable repository files, you may need to stash your changes and then reapply them after pulling:

cd $MODDABLE: **_Should be %MODDABLE%_**
git stash push
git pull
git stash pop

andycarle, 2022-04-08:
Hi @louisvangeldrop,

Thank you for the suggestion.

Those particular instructions are written for updating with [Git Bash](https://gitforwindows.org/) ("For instance, using Git Bash:"), which uses the bash-like `$` syntax for environment variables rather than the Windows `%%`. 

That said, I agree that it is a little confusing that parts of the Windows instructions assume Git Bash and parts assume CMD. At the time they were written, Git integration with CMD was somewhat uncommon — now I think more people are using that model than using Git Bash. 

I will take a pass at some point and drop the Git Bash pieces from the install/update flows.

Cheers!
&nbsp;- Andy

louisvangeldrop, 2022-04-08:
You are welcome.

closed 2022-06-03 by phoddie