Moddable-OpenSource/moddable #898: Supplementary-plane characters in non-unicode regular expression group names are incorrectly rejected

gibson042, 2022-04-12:
**Environment:** XS 11.6.0 32 4

**Description**
Independent of the `u` flag, regular expression group names should be allowed to contain `\u{…}` escape sequences and arbitrary IdentifierName characters. But XS incorrectly restricts behavior in absence of the flag to reject both `\u{…}` escapes and characters outside the Basic Multilingual Plane.

**Steps to Reproduce**
Execute the tests at https://github.com/tc39/test262/blob/a226601d12dac85fb779f7396d6bc28aa3038963/test/built-ins/RegExp/named-groups/non-unicode-property-names-valid.js .
Notable excerpts:
- `/(?<the𝟚>the)/`
- `/(?<the\u{1d7da}>the)/`
- `/(?<the\ud835\udfda>the)/`

**Expected behavior**
Per [GroupName](https://tc39.es/ecma262/multipage/text-processing.html#prod-GroupName) expansions specifically including <tt>`\` RegExpUnicodeEscapeSequence<sub>[**+UnicodeMode**]</sub></tt> and <tt><sub>[~UnicodeMode]</sub> **UnicodeLeadSurrogate UnicodeTrailSurrogate**</tt>, `\u{…}` escape sequences and supplementary-plane characters should be allowed in group names of regular expressions with or without a `u` flag (but note that per [pattern Early Errors](https://tc39.es/ecma262/multipage/text-processing.html#sec-patterns-static-semantics-early-errors), the resulting group name must start with a Unicode "ID_Start" code point and consist only of "ID_Continue" code points, excluding e.g. `/(?<\u{1D306}>.)/`).

**Actual behavior**
Non-unicode regular expressions with group names including `\u{…}` escape sequences and/or non-BMP characters are rejected.

phoddie, 2022-05-31:
FWIW – it looks like this may work correctly with CESU-8 support is enabled. When building `xst` with CESU-8 enabled (using our internal development version of XS 13) , these tests pass.

phoddie, 2022-08-22:
I just ran the tests linked to above. No errors were reported. It isn't impossible that it was fixed in XS In the last few months. Can you check to see if the problem is resolved for you with the latest XS? Thank you.

gibson042, 2022-08-23:
Looks good, thanks!

closed 2022-08-23 by gibson042