Moddable-OpenSource/moddable #899: XS appears to incorrectly ignore comma at end-of-line

gibson042, 2022-04-12:
**Environment:** XS 11.6.0 32 4

**Description**
XS appears to incorrectly allow automatic semicolon insertion to terminate a statement that includes a comma before a [LineTerminator](https://tc39.es/ecma262/multipage/ecmascript-language-lexical-grammar.html#prod-LineTerminator).

**Steps to Reproduce**
Execute the following syntactically incorrect code:
```js
const a = 0,
const b = 1;
print(a, b);
```

**Expected behavior**
No code is evaluated and a syntax error is reported.

**Actual behavior**
`a` and `b` bindings are created and the `print` reports corresponding values.
```
0 1
```

phoddie, 2022-05-31:
FYI – XS also accepts this line, so the issue isn't related to automatic semicolon insertion.

```js
const a = 0, ; const b = 1;
```

This is expected to be fixed in XS13.

phoddie, 2022-06-06:
Fixing this broken our build because ssl_handshake.js used commas at the end of a few lines where it should have used semicolons. That's fixed now too.

closed 2022-06-30 by phoddie