Moddable-OpenSource/moddable #907: WeakMap crashes with exception when key is a preloaded Class

cmidgley, 2022-05-19:
**Build environment:** Linux
**Target device:** ESP32 

**Description**
`WeakMap` crashes with exception when a preloaded module has a class that is added as the key object. 

The failure is in `xsMapSet.js` with an exception [here](https://github.com/Moddable-OpenSource/moddable/blob/a41e6dbd904959983bf55323dcd49ba2cc883f87/xs/sources/xsMapSet.c#L1278). 

**Steps to Reproduce**
Small fragment that reproduces the failure in [this gist](https://gist.github.com/cmidgley/308fb4211cb8aed093faca94cd225372).

phoddie, 2022-05-19:
It appears the problem happens when adding any truly immutable objects (those stored in flash/ROM/read-only memory) to a `WeakMap`.

phoddie, 2022-05-30:
The crash has been fixed for keys in ROM for all methods of `WeakMap` and `WeakSet`. On `WeakMap`, `set` and `delete` throw for keys in ROM. Since keys in ROM can never be collected, it is kind of meaningless to put them in a `WeakMap` or `WeakSet` – though there are cases where it can be useful.

For the moment, this is change we're comfortable making to `WeakMap` and `WeakSet`. It is possible to support keys in ROM but it would require an additional checks and a separate table. That might make sense down the road, but we'd like to better understand how necessary it is.

Meanwhile, the fixes make it possible to create a polyfill that allows ROM keys to work on a `WeakMap`. Here's an example of that. Getting the polyfill right for all cases is a little tricky, so this might not be 100% yet. Working through the polyfill is a good step to understanding what an implementation of this in XS might eventually need to look like.

```js
globalThis.WeakMap = class WeakMapWithROMKeys extends WeakMap {
	#map;
	
	delete(key) {
		if (this.#map?.delete(key))
			return true;

		try {
			return super.delete(key);
		}
		catch {
		}
		return false;
	}
	get(key) {
		return super.get(key) ?? this.#map?.get(key)
	}
	has(key) {
		if (super.has(key))
			return true;

		return this.#map?.has(key) ? true : false;
	}
	set(key, value) {
		try {
			super.set(key, value);
		}
		catch {
			const type = typeof key;
			if (("function" === type) || ("object" === type)) {
				this.#map ??= new Map;
				this.#map.set(key, value);
			}
		}
	}
}
```

closed 2022-05-31 by cmidgley