Moddable-OpenSource/moddable #910: Is it possible to use a node library on my application?

fernandotetu, 2022-05-25:
I would like to ask about the possibility of using a node library on my project. I'm assuming that my hardware there are capabilities to support it.

In particular, I would like to use [cron library](https://www.npmjs.com/package/cron). My intention is to execute an action every day at a specific time. For that, I have downloaded the library and his dependency and point for the node_module folder on my manifest file (I try diferent entry and paths). However, I always received a message: "import CronJob not found!"

I would really appreciate it if anybody could help me to use node libraries with Moddable

wilberforce, 2022-05-26:
post your manifest file

phoddie, 2022-05-26:
As noted in the New Issue template:

> Questions opened as an issue will be converted to a Discussion.

Moving to Discussion.