Moddable-OpenSource/moddable #913: Windows mcconfig tooling build fails due to new ShareArrayBuffer support

HipsterBrown, 2022-06-01:
**Build environment:** Windows

**Description**
Attempting to build mcconfig on Windows fails due to `error C2036: 'void *' : unknown size`. I found a [related StackOverflow](https://stackoverflow.com/a/37460703) to further describe the potential issue.

[Build logs](https://github.com/HipsterBrown/moddable-prebuild/runs/6694631253?check_suite_focus=true#step:7:242)
Trace leads back to [sharedArrayBuffer in `xsString.c](https://github.com/Moddable-OpenSource/moddable/commit/28e1bc72b4e20bbeb0f1ced36289336b140ae64b#diff-ddaf16d47c3d3cd43a46737c8216ee6a8a8be1a5c9b8a38c7ddde35478bcb6adR406)

**Steps to Reproduce**
1. Clone latest commit moddable repo to Windows environment
2. Change directories to `moddable/build/makefiles/win/`
3. Execute (run) `build.bat`

**Expected behavior**
Successfully build mcconfig and other tools

**Images**
[image]

phoddie, 2022-06-01:
Thank you for reporting this. The Microsoft compiler is an interesting outlier – complaining about things that GCC & Clang let slide.

It should be enough to cast the `address`:

```c
c_memcpy(string, offset + (arrayBuffer ? arrayBuffer->value.arrayBuffer.address : (txByte *)sharedArrayBuffer->value.host.data), outLength);
```

We'll get an update out after testing....

phoddie, 2022-06-01:
Fingers crossed.

HipsterBrown, 2022-06-02:
Appears to be fixed! 🚀👏

closed 2022-06-02 by HipsterBrown