Moddable-OpenSource/moddable #914: Ordinary Object constructor function does not have a `name` property

dtex, 2022-06-02:
**Build environment:** macOS
**Target device:** esp

**Description**
I'm a beginner when it comes to reading 262, but I think I *might* have found a conformance issue. Per section 18, "Every built-in [function object](https://262.ecma-international.org/12.0/#function-object), including constructors, has a "name" property whose value is a String." In XS this property is undefined when looking at an ordinary object instance.

**Steps to Reproduce**
```js
let x = { y:0 };
trace(x.constructor.name);

// logs undefined
```

**Expected behavior**
As seen in V8
```js
let x = { y:0 };
trace(x.constructor.name);

// logs "Object"
```

phoddie, 2022-06-02:
@dtex, thanks for the report. Reading ECMA-262 is never easy. I've been trying for years and still feel like a beginner.

**Short answer**: This is an intentional variance from strict conformance.

**Longer answer**: For embedded targets, by default the Moddable SDK builds XS with reduced conformance in a number of areas to improve performance and reduce code size. You will recall, for example, that top-level-await is disabled by default for the `main` module for these reasons.

The [runtime](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20Conformance.md#runtime-models) section of the XS Conformance document talks about this in a bit more detail. Of relevance to your report, it notes:

> ...the XS linker can... remove length and name properties from functions

Because of this, no built-in functions have a `name` or `length` property. For example, `Object.freeze.name` is `undefined` too. On a device with just 1 MB of flash like the ESP8266, removing all those `name` and `length` makes a meaningful difference for projects tight on space, and they are rarely missed. 

The conformance document is a bit misleading in suggesting that this behavior is optional. If preload is used, the XS linker does remove `name` and `length` from built-in functions. For normal projects, disabling preload isn't a viable option because of the increased RAM use. One project that does keep `name` and `length` is test262 test runner for embedded devices.  For that, we need maximum conformance so we don't preload any modules.

dtex, 2022-06-02:
That makes perfect sense and thank you for taking the time to explain it to me.