Moddable-OpenSource/moddable #916: TLS handshake failing on using Application Layer Protocol Negotiation(ALPN)

barianand7, 2022-06-06:
## TLS handshake failing on using Application Layer Protocol Negotiation(ALPN)

#### Environment :
- ESP-IDF version: v4.4 otafix
- Moddable version: [b18d0b0](https://github.com/Moddable-OpenSource/moddable/commit/b18d0b012134ef16679b95adb0aa505e4dfb76f7)
- Build System: mcconfig
- Compiler version: Microsoft visual studio C++ framework 
- Operating System: Windows 10

#### Problem Description :
TLS handshake fails on using application layer protocol negotiation(ALPN) as part of the TLS extension. 

#### Expected Behavior:
TLS handshake should work fine even with the ALPN extension enabled.

#### Actual Behavior :
After enabling ALPN extension, moddable breakdown on this point `if (typeof ext == 'string') ext = ext.split(':');`

#### Steps to reproduce :
The problem can be reproduced by using the `httpgets` moddable example application with below modifications mentioned in the Code to reproduce this issue section:
1) Configure the code as per following where the problem is reproduced.
`		secure: {
			protocolVersion: 0x303,
			applicationLayerProtocolNegotiation: "mqtt"				// Adding ALPN 
		}`
2) Add the certificate `"$(MODULES)/crypt/data/ca222"` additionally if it is reported an failure on xsbug.
3) Build and Run the `httpsgets` moddable example application in desktop simulator (mcconfig -d -m -p win)
4) Monitor the failure on xsbug console logs

#### Code to reproduce this issue :

[image]
[image]

#### Analysis : 
`const ext = session.options[i];`
In SSL_Handshake.js file `ext` is defined as const.
Code `if (typeof ext == 'string') ext = ext.split(':');` has been break down where `const ext` variable tried to update with new value. which causes break issue to establish handshaking

[image]

#### Proposed Solution :

Declare the `const ext` variable with `let ext` instead of `const`  in `ssl_handshake.js` module.
(Problem were not found before Github [commit ](https://github.com/Moddable-OpenSource/moddable/commit/efe5107169243d24c8ec85a9e1523fcba79ec527)).

[image]
[image]

phoddie, 2022-06-11:
Thank you for the detailed report. We'll resolve this in our next update.

phoddie, 2022-06-16:
This issue has been resolved by [this commit](https://github.com/Moddable-OpenSource/moddable/commit/00872b1f108b96d422454f6fcd4358410f32bc09). The commit adds support for binary and UTF-8 protocol names. It also changes the way multiple protocols are passed (now an array rather than a single string with separator). The documentation has been updated to reflect these changes.

barianand7, 2022-06-27:
Issue has been verified and seems resolved with [New commit](https://github.com/Moddable-OpenSource/moddable/commit/00872b1f108b96d422454f6fcd4358410f32bc09)

[image]

closed 2022-06-27 by barianand7