Moddable-OpenSource/moddable #917: BLE examples crash on build for target esp32/esp32s3

diogoviannaaraujo, 2022-06-22:
**Build environment:** macOS ARM
**Target device:** esp32-s3

**Description**
Build fails when trying to use bluetooth on the esp32-s3 giving this error:
`#define BT_CONTROLLER_INIT_CONFIG_DEFAULT() {0}; _Static_assert(0, "please enable bluetooth in menuconfig to use bt.h");`

**Steps to Reproduce**
1. Open any BLE example and run mcconfig -d -m -p esp32/esp32s3

**Images**
[image]

Been trying to find where bluetooth is enabled on sdkconfig for esp32, found this:
https://github.com/Moddable-OpenSource/moddable/blob/e11fa10d5c9804a6ddbad4941ed8645d8a9a29c8/tools/mcmanifest.js#L223
still unsure why is not working on the esp32-s3

andycarle, 2022-06-22:
@diogoviannaaraujo I have a fix for this committed — it will be in our next open source push. 

In the meantime, the fix would be easy for you to apply locally. In [$MODDABLE/build/devices/esp32/xsProj-esp32s3/sdkconfig.defaults](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/xsProj-esp32s3/sdkconfig.defaults), replace [lines 184-188](https://github.com/Moddable-OpenSource/moddable/blob/e11fa10d5c9804a6ddbad4941ed8645d8a9a29c8/build/devices/esp32/xsProj-esp32s3/sdkconfig.defaults#L184-L188) with:

```
#
# Bluetooth
#
CONFIG_BT_ENABLED=n
CONFIG_BT_BLUEDROID_ENABLED=n

CONFIG_BT_NIMBLE_ENABLED=n
CONFIG_BT_NIMBLE_MAX_CONNECTIONS=1
CONFIG_BT_NIMBLE_MAX_BONDS=3
CONFIG_BT_NIMBLE_MAX_CCCDS=8
CONFIG_BT_NIMBLE_L2CAP_COC_MAX_NUM=0
CONFIG_BT_NIMBLE_PINNED_TO_CORE_0=y
CONFIG_BT_NIMBLE_PINNED_TO_CORE_1=n
CONFIG_BT_NIMBLE_PINNED_TO_CORE=0
CONFIG_BT_NIMBLE_TASK_STACK_SIZE=4096
CONFIG_BT_NIMBLE_ROLE_CENTRAL=n
CONFIG_BT_NIMBLE_ROLE_PERIPHERAL=n
CONFIG_BT_NIMBLE_ROLE_BROADCASTER=n
CONFIG_BT_NIMBLE_ROLE_OBSERVER=n
CONFIG_BT_NIMBLE_NVS_PERSIST=y
CONFIG_BT_NIMBLE_SM_LEGACY=y
CONFIG_BT_NIMBLE_SM_SC=y
CONFIG_BT_NIMBLE_DEBUG=n
CONFIG_BT_NIMBLE_SM_SC_DEBUG_KEYS=n
CONFIG_BT_NIMBLE_ATT_PREFERRED_MTU=256
CONFIG_BT_NIMBLE_MESH=n
CONFIG_BT_NIMBLE_CRYPTO_STACK_MBEDTLS=n
```

phoddie, 2022-06-30:
Fixed!