Moddable-OpenSource/moddable #932: Look at recommendation to run brew update, which can take multiple minutes depending on state of host computer

gingerbeardman, 2022-08-26:
(Following on from [twitter thread](https://twitter.com/gingerbeardman/status/1563212587499524096))

**Describe**
When doing setup/install, during the ESP-IDF phase, it is advised to run `brew update`. Depending on the state of the host computer (how many packages are installed using brew, when `brew update` was last run, network and cpu speeds, etc) the `brew update` process can take tens of minutes to complete - for me it took over 30 minutes on a 10-core M1 Pro MBP running macOS 12.5.1.

See: Step 5 in [documentation/devices/esp32.md](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#mac-instructions) (emphasis mine)

> 5. *Update homebrew* and then install Python, cmake, ninja, the pip package management system, and pyserial. Also run a brew upgrade on those packages, in case you already had older versions installed:
> ```
> brew update
> brew install python3 cmake ninja dfu-util
> brew upgrade python3 cmake ninja dfu-util
> sudo easy_install pip
> pip install pyserial
> ```

Note: be wary of the differences between
1. `brew update`
   - updates homebrew itself
   - updates version information for all packages installed
   - can take many minutes depending on current config of host computer)
2. `brew upgrade`
   - downloads and installs newer package binaries

**Why do you think this feature would be useful?**
- quicker setup/install would reduce friction and make onboarding more pleasurable
- reduce the number of users who drop-off at the install phase

and
- is a mass `brew update` truly needed?
- is it desirable to allow `brew update` to add a speed bump to your install process?

**Describe alternatives you've considered**
- update version info of only ESP-IDF dependencies 
- use an alternative package management approach that does not have adverse runtime surprises
  - manually
  - nix
  - macports
  - etc.

andycarle, 2022-08-26:
Hi @gingerbeardman — thank you for the report. I'm sorry to hear that this portion of the install process was misbehaving for you.

It's curious (and unfortunate) that `brew update` took so long to complete on your machine. My understanding is that `brew update` is intended to be a very fast operation. On a quick survey of macOS machines we have available on the team, it took somewhere in the range of 10 to 60 seconds to complete. That said, I see a variety of GitHub issues and StackOverflow posts out there with people complaining about slow `brew update` executions. (e.g. https://github.com/Homebrew/brew/issues/4755 and https://github.com/Homebrew/brew/issues/13224) And anecdotally I've heard a couple of times that early `brew` installs on M1 hardware were particularly broken and may need to be re-installed now that the kinks have been worked out.

I will consider some alternatives here, including those that you've listed. Unfortunately, it is not possible, as far as I know, to scope the execution of `brew update`. That is, it does not take any arguments to restrict the update to specific packages/taps in the way that `brew upgrade` does.

MacPorts would be fine. Indeed, if you have MacPorts installed it would be just as good to do `port install cmake ninja dfu-util`. 

Brew and MacPorts are the two options suggested by Espressif in their [ESP-IDF setup instructions](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#for-macos-users), so I'm inclined to agree that we should include that option in our ESP32 guide as well. Perhaps it would suffice to go with the ESP-IDF language and explain how to do this with either `brew` or `port` and also link to direct installers.

gingerbeardman, 2022-08-26:
I'll leave the final decision to you, no worries.

It is indeed concerning and perhaps this is something I can take up with brew. My M1 laptop is only a couple of months old so brew is quite fresh, I don't use it much.

I wish I could turn back time and profile the commands a little more, issuing them one by one. On closer inspection I wonder if I pasted all the lines at once, and thus could it have been something other than the brew update taking a long time? I'll give it some thought.

It would be great if before each set of commands expectations could be set as to the time taken. I wondered if it was normal for the ESP-IDF install to take 30+ minutes, or whether that was only with brew? So I did `port install cmake ninja dfu-util` and it completed in a couple of minutes.