Moddable-OpenSource/moddable #940: Executing "mcrun" command leading to NMake fatal error

nickthiru, 2022-10-05:
**Build environment:** Windows 11
**Target device:** NodeMCU ESP32

**Description**
I recently got started with the book "IoT Development for ESP32 and ESP8266 with JavaScript" and am facing an issue. I am able to install a host (from the exercises in the book) using `mcconfig`. However, when executing `mcrun`  to install one of the examples, I'm seeing the error shown in the attached screenshot, under Images:

**Steps to Reproduce**
1. Install a host using this command: `mcconfig -d -m -p esp32/nodemcu`
2. Install a corresponding exercise using this command: `mcrun -d -m -p esp32/nodemcu`
3. See error

**Expected behavior**
I expected the programs (the exercises from the book) to successfully run after I have installed the host.

**Images**
[image]

**Other information**
It was working a couple of weeks ago and then yesterday, I started experiencing this issue. I even re-setup everything twice, including restoring my system to an earlier restore point (I believe it was to a time before I started with the first-time setup). Also note that I hadn't worked on the exercises for about a week and something must have happened in the interim. I did experience the problem with a particular nodemcu (but it was working initially); basically the USB connector fell off the board (but the problem was there before this happened). I purchased a new nodemcu today and have been facing the reported issue.

andycarle, 2022-10-05:
Hi @nickthiru,

I'm sorry that you're seeing issues with `mcrun`. Let's see if we can get you back up and running again.

The error that you're seeing there indicates that the `helloworld` mod is failing to install correctly. Most likely it is either arriving to the device corrupted or the device does not have a proper location to install the mod. 

A couple of things to try:

- Believe it or not, we often see this error when a bad/noisy USB cable is used to install mods. Try a different cable if you have another one available. 
- Erase the device entirely and start over by installing the host and mod. To do the erase, use the idf.py tool that comes with the ESP-IDF: `python %IDF_PATH%\components\esptool_py\esptool\esptool.py erase_flash`. Then use `mcconfig` to install `ch1-gettingstarted\host` and `mcrun` to install `ch1-gettingstarted\helloworld`.

Please let me know if either of those ideas helps. If neither works, we can try adjusting the baud rate used by xsbug to communicate with your nodemcu board.

nickthiru, 2022-10-07:
Hello Andy, it's a pleasure to hear from you. thank you very much for the prompt and detailed reply. Soon after I reported this issue, I was in the process of beginning to test everything out on my laptop (as opposed to the workstation where I experienced the issue) when my laptop started acting up for some strange reason; it was crashing when I tried to install ms visual studio components. and it started to crash at other times soon after that. I am now resetting my laptop before I try it again. I will also try your solutions out on my workstation. This was the reason I haven't been able to get back to you on my results. I just wanted to report back to you asap on the status. Thanks again. I'll be in touch

nickthiru, 2022-10-10:
Hi Andy,

I have reset my laptop and tried changing to a brand new USB cable but I'm still unable to run `mcrun` successfully. I'm still seeing the "fatal error" from "serial2xsbug" (on both my workstation and laptop). I have also witnessed the following as well (even before trying a new USB cable):

1) I have seen a virus threat warning in the (ESP-IDF) command prompt after running `mcrun` on a couple of occasions. I saw the virus threat in Windows Defender app as well. However, the warning goes away if I try running `mcrun` again, whether I run the Windows Defender scan (which also "removes" the virus threat warning, sometimes on it's own) or not.
[image]

2) I have seen this error as well a couple of times as well (notice it happens on `mcconfig`):
[image]

Finally, I erased the device completely like you suggested but when I try to execute `mcconfig`, I'm seeing the error highlighted in 2), above.

andycarle, 2022-10-11:
Hi @nickthiru,

I'm going to bet that the two problems you've identified here are related. Most likely, Microsoft Defender is quarantining `tools.exe` and preventing it from running as part of `mcconfig`.

The Microsoft Defender trigger is a false positive and is (very) annoying. We've reported it to Microsoft a couple of times now and had our executables white-listed, but it seems to crop up again for some users every time anything in the build changes. 

Usually when that detection is triggered, Defender will quarantine the impacted executable. If you go to `%MODDABLE%\build\bin\win\release` and see what's in there, you should be able to tell. Mine looks like:

```console
 Directory of C:\moddable\build\bin\win\release

10/03/2022  04:04 PM    <DIR>          .
10/03/2022  04:03 PM    <DIR>          ..
10/03/2022  04:03 PM                28 bles2gatt.bat
10/03/2022  04:03 PM                28 buildclut.bat
10/03/2022  04:03 PM                22 cdv.bat
10/03/2022  04:03 PM                34 colorcellencode.bat
10/03/2022  04:03 PM                30 compressbmf.bat
10/03/2022  04:03 PM                27 image2cs.bat
10/03/2022  04:03 PM                27 mcbundle.bat
10/03/2022  04:03 PM                27 mcconfig.bat
10/03/2022  04:03 PM                26 mclocal.bat
10/03/2022  04:03 PM                24 mcrez.bat
10/03/2022  04:03 PM                24 mcrun.bat
10/03/2022  04:04 PM         2,292,736 mcsim.exe
10/03/2022  04:03 PM                30 nodered2mcu.bat
10/03/2022  04:03 PM                26 png2bmp.bat
10/03/2022  04:03 PM                29 rle4encode.bat
10/03/2022  04:03 PM           169,984 serial2xsbug.exe
10/03/2022  04:03 PM         1,998,848 simulator.exe
10/04/2022  12:12 AM         1,393,664 tools.exe
10/03/2022  04:03 PM                27 wav2maud.bat
10/03/2022  04:04 PM         2,826,240 xsbug.exe
10/03/2022  04:03 PM           344,064 xsc.exe
10/03/2022  04:03 PM           166,912 xsid.exe
10/03/2022  04:03 PM           911,360 xsl.exe
```

If you're missing any of those executables (most obviously `tools.exe`) you'll definitely have a problem. 

A couple of things to try:
1. You can exclude the Moddable SDK build directory from Defender by following [these instructions](https://support.microsoft.com/en-us/windows/add-an-exclusion-to-windows-security-811816c0-4dfd-af4a-47e4-c301afe13b26). 
2. I would recommend completely deleting the `%MODDABLE%\build\bin` and `%MODDABLE%\build\tmp` directories and then rebuilding the Moddable SDK tools. 
3. After doing that, verify that the exclusion has stuck and perhaps explicitly exclude the newly-built `%MODDABLE%\build\bin\win\release` directory from Defender.

Please give those a shot and let me know how it goes.

nickthiru, 2022-10-13:
@andycarle Hey there,

Thank you very much for sharing the details of your directory; I did go into the folder earlier to confirm that tools.exe was indeed missing. And thanks to your directory details, I also saw that xsl.exe is missing. I can now recall seeing an error, in the past, reporting that xsl.exe was missing (I'm not sure if tools.exe was also missing at that time, although the error did not report it specifically).

I followed your instructions and managed to get back on track i.e. get mcconfig working. However, I hit another issue, which is something else that I have seen, in the past, when I wasn't having a problem with `mcconfig`. Kindly note the error message (2) at the end of the console:
[image]

Additional Questions:
1. After installing the host, I pressed Ctrl-C twice, as seen in (1) in the above screenshot. I'm assuming this is fine and has no bearing on the issue. Am I correct in also assuming that once the host has been installed the process that `mcconfig` started is still running in that console (hence, the need to terminate the process with `Ctrl-C`)?
2. Why did `mcconfig` execute successfully and `mcrun` didn't? Is it because `mcrun` doesn't use `tools.exe` and/or `xsl.exe`
3. The following commands only need to be issued once, am I correct? And can you kindly confirm which console are they supposed to be run in: normal command prompt, x86 Native Tools command prompt, or ESP-IDF command prompt?
```
> cd %IDF_PATH%
> install.bat
```
Thank you.

andycarle, 2022-10-14:
> After installing the host, I pressed Ctrl-C twice, as seen in (1) in the above screenshot. I'm assuming this is fine and has no bearing on the issue. Am I correct in also assuming that once the host [...]

Correct.

> Why did mcconfig execute successfully and mcrun didn't? Is it because mcrun doesn't use tools.exe and/or xsl.exe

The process that the two tools use is quite different. `mcconfig` rewrites the entire flash of the device using ESP-IDF utilities. `mcrun` communicates with the Moddable software that is now running on the device to install the mod. It appears that that communication (or the subsequent storage of the mod in flash) is what is failing.

> The following commands only need to be issued once, am I correct? And can you kindly confirm which console are they supposed to be run in: normal command prompt, x86 Native Tools command prompt, or ESP-IDF command prompt?

Correct. It should be safe to run `install.bat` from either command prompt, as long as you restart after. But this is not obviously related to your problem.

I need to contemplate what the issue here might be and try a couple of things. I'll try to get back to you late this afternoon with a couple of experiments to try out.

andycarle, 2022-10-17:
Hi @nickthiru,

I've been thinking more about this issue. I have one quick thing to try. Could you please try building and running a debug version of the `helloworld` example in the Moddable SDK?

You can do this with:

```console
cd %MODDABLE%\examples\helloworld
mcconfig -d -m -p esp32/nodemcu
```

If everything goes as expected, that should:

1. Build the `helloworld` app
2. Flash the app to your nodemcu device
3. Open the xsbug debugger to connect to the app on the device
4. Run the app on the nodemcu, triggering a break in xsbug on the `debugger` statement in the `helloworld` source code

If that doesn't work correctly, we definitely have something with communication between your machine and the nodemcu board, which we can work to debug next.

andycarle, 2022-10-22:
That's correct, yes. If everything worked correctly, you should end up in xsbug looking like this:

[image]

I'm just trying to verify that the communication between serial2xsbug and your device is working at all.

nickthiru, 2022-10-22:
Yup, noted. Yes, this is exactly what I see in the end, after executing `mcconfig`. I usually also hit the "run" button in the toolbar at the top just to be sure. I'm assuming you will get back to me after further analysis?

Thanks.

andycarle, 2022-10-25:
Hi @nickthiru,

Thank you for your patience. 

Today, my colleague added some additional diagnostic output to `mcrun` to help us narrow down the cause of the problem you're seeing. I would appreciate it if you could try again with the new version of `mcrun` and report back on what error you see.

To update and re-run, please:

1. Delete the `%MODDABLE%\build\bin` and `%MODDABLE%\build\tmp` directories
2. Pull updated sources from GitHub (`cd %MODDABLE% & git pull`)
3. Rebuild the Moddable SDK tools (`cd %MODDABLE%\build\makefiles\win & build`)
4. Redeploy the host (`cd %USERPROFILE%\Projects\iot-product-dev-book\ch1-gettingstarted\host & mcconfig -d -m -p esp32/nodemcu` and the mod (`cd %USERPROFILE%\Projects\iot-product-dev-book\ch1-gettingstarted\helloworld & mcrun -d -m -p esp32/nodemcu`

You will presumably still see an error, but it should be a new error code (not the `65535` you were seeing before).

Thank you!

nickthiru, 2022-10-26:
Hi @andycarle 

No worries; happy to help narrow down the issue.

Please see the attached screenshot of the console output and xsbug window below. For some reason, it "seems to have worked"? Because the former error messages don't seem to be appearing and the console window is showing "complete" status for installing the mod, and xsbug launched with the `example.js` file loaded.

[image]
[image]

I'm going to try installing the host and mod for the "connecting to wifi" part of chapter 3, Networking, and see what happens. That's where I got stuck initially (after having success completing the "hello world" example) and raised the bug report.

nickthiru, 2022-10-26:
This looks promising!

[image]
[image]

I will continue working through the Networking chapter and test the connection while I wait for your reply. I am very curious as to why it "seems" to be working now. Please note that I have successfully been using Mongoose OS with my MCU while we worked through the issues that I was facing with Moddable. I'm not sure if that had any bearing on what, now, seems to be successful executions of both the host and the mod, and of more than one exercise in the book. If, indeed, you concur that the issue has been fixed, I am very much curious as to what fixed it, because you were expecting `mcrun `to fail and display error messages, if I am not mistaken, right?

andycarle, 2022-10-26:
Interesting! Yes, this all looks correct. I'm glad to see that everything is working for you now!

> I am very much curious as to what fixed it, because you were expecting mcrun to fail and display error messages, if I am not mistaken, right?

Correct, and I share your curiosity. I can think of a few possibilities:

- Perhaps somehow the `erase_flash` command that we tried earlier in this process did not completely work and the region of flash that we use for installing the mod was still corrupted. It seems plausible that your work with Mongoose OS would have done a full erase that more properly worked at some point, leading to your later success with our stack here.
- This could happen if somehow you had mismatched versions of the host and the mod. For instance, if you had built the host on an older version of the Moddable SDK and then never fully cleaned it. That doesn't seem likely given our debugging steps above, but it is possible.
- While also not likely, perhaps something about your USB-to-serial driver state changed between attempts here. We've seen something as simple as rebooting the PC help in these situations.

If you find yourself back in the broken state, we can resume our investigation. But, fingers crossed that that doesn't happen!

nickthiru, 2022-10-27:
Glad to hear you concur that everything is up-to-mark now. And thanks for mentioning the possible fixes; it will help with troubleshooting in  future. If there are any issues, I will report back to you.

Thank you very much for your kind assistance @andycarle and team. I look forward to working with Moddable and its ecosystem going forward.

P.S. Just a sidenote. I am trying to work with AWS IoT and it was a whole lot easier to interface with AWS IoT using Mongoose OS. After I got successful WiFi  connection yesterday I began investigating how I will be able to connect to AWS IoT while working with the Moddable SDK. There wasn't a whole lot of search results in Google addressing this matter. I was just wondering, since I'm still new to the area, whether this is something that would have already been done i.e. ESP32 + Moddable + AWS IoT, and if anyone can point me in the right direction, if some documentation exists on this topic. I would be glad to come up with an article or such, if I have to do it from scratch, so that others can benefit. Thanks.

andycarle, 2022-10-27:
> Thank you very much for your kind assistance @andycarle and team. I look forward to working with Moddable and its ecosystem going forward.

You're welcome! We're glad to have you here.

> P.S. Just a sidenote. I am trying to work with AWS IoT and it was a whole lot easier to interface with AWS IoT using Mongoose OS. After I got successful WiFi connection yesterday I began [...]

We do have a fair bit of experience working with AWS and AWS IoT. Our networking modules (HTTP, TLS, MQTT, etc.) are well-suited for the task—it's mostly a matter of gluing the pieces together to achieve your goals. But, to my knowledge we do not have any sort of overview documentation for working with AWS in the Moddable SDK.

My recommendation would be to open up a new [GitHub discussion](https://github.com/Moddable-OpenSource/moddable/discussions) with a bit more information about what you're hoping to accomplish. Then we should be able to get the right people to help you out.

Cheers!

nickthiru, 2022-10-27:
Thank you very much for the info on working AWS IoT. I figured this would be a non-issue and it's mostly about gluing the pieces together, like you said. I will open a new GitHub discussion for this. Thanks again!

closed 2022-10-27 by nickthiru