Moddable-OpenSource/moddable #948: LED not blinking on ESP32C3

louisvangeldrop, 2022-10-24:
**Build environment:** Windows,
**Target device:** ESP32-C3 NodeMCU devkit

**Description**
LED not blinking using the Cold White LED-pin:19 running the Blink example from the moddable\ecamples library

[image]

Arduino example:

[image]

The Arduino example shows a blinking LED.

mkellner, 2022-10-25:
Thank you for the bug report.

The ESP32C3 defaults to using pins 18 and 19 for USB.
Setting the function of the pin to GPIO wasn't enough, but resetting the pin before setting the function clears the USB feature.

I have also added some targets for the AI Thinker ESP32 C3-32S-Kit. `esp32/c3_32s_kit` is for the 4MB version and `esp32/c3_32s_kit_2m` is for the 2MB version.

louisvangeldrop, 2022-10-25:
@mkellner 
Thank you for the quick response and the fix. It works now.

louisvangeldrop, 2022-10-25:
I have tried the io-blink sample with this board and do get the following error:

[image]

phoddie, 2022-10-25:
This appears to be an unrelated issue. If that is correct, please close this issue and open a new one. Thank you.

closed 2022-11-09 by phoddie