Moddable-OpenSource/moddable #950: nodered2mcu fails to build w/ configuration nodes

ralphwetzel, 2022-10-26:
Configuration nodes don't carry the `wires` property.
Thus `nodered2mcu` trows a TypeError @ line 166:

https://github.com/Moddable-OpenSource/moddable/blob/c635af79872c0bc0e5df644c3b664caa2a8f7710/tools/nodered2mcu.js#L161-L180
Can be fixed w/ `config.wires?.forEach(output => { `

It throws another TypeError @ line 237.

https://github.com/Moddable-OpenSource/moddable/blob/c635af79872c0bc0e5df644c3b664caa2a8f7710/tools/nodered2mcu.js#L236-L238
Can be fixed w/ `for (let i = 0; i < config.wires?.length; i++) `

**This** TypeError fails to be reported due to a ReferenceError in the `catch` clause - as `name` is not defined.

https://github.com/Moddable-OpenSource/moddable/blob/c635af79872c0bc0e5df644c3b664caa2a8f7710/tools/nodered2mcu.js#L283-L286
`name` should become `c.name`.

phoddie, 2022-10-26:
Funny. I fixed a similar issue last night (not committed yet) when working on basic group support. I'd like to make sure all cases are covered before committing the changes. Would you share a flow that shows the configuration node problem? Thank you!

ralphwetzel, 2022-10-26:
Here is a test case that shows the config node issue:

[image]

``` json
[
    {
        "id": "da4aa1c13327d032",
        "type": "tab",
        "label": "Flow 8",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": true
        }
    },
    {
        "id": "e4e47afb90a8cd59",
        "type": "mqtt in",
        "z": "da4aa1c13327d032",
        "name": "",
        "topic": "test",
        "qos": "2",
        "datatype": "auto-detect",
        "broker": "31795cd0c1e59be4",
        "nl": false,
        "rap": true,
        "rh": 0,
        "inputs": 0,
        "_mcu": {
            "mcu": true
        },
        "x": 630,
        "y": 380,
        "wires": [
            [
                "748774df738b3310"
            ]
        ]
    },
    {
        "id": "58bb176115cff996",
        "type": "mqtt out",
        "z": "da4aa1c13327d032",
        "name": "",
        "topic": "test",
        "qos": "",
        "retain": "",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "31795cd0c1e59be4",
        "_mcu": {
            "mcu": true
        },
        "x": 490,
        "y": 380,
        "wires": []
    },
    {
        "id": "7ed461d9e91beb0f",
        "type": "inject",
        "z": "da4aa1c13327d032",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "_mcu": {
            "mcu": true
        },
        "x": 360,
        "y": 380,
        "wires": [
            [
                "58bb176115cff996"
            ]
        ]
    },
    {
        "id": "748774df738b3310",
        "type": "debug",
        "z": "da4aa1c13327d032",
        "name": "debug 39",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": {
            "mcu": true
        },
        "x": 760,
        "y": 380,
        "wires": []
    },
    {
        "id": "31795cd0c1e59be4",
        "type": "mqtt-broker",
        "name": "",
        "broker": "localhost",
        "port": "1883",
        "clientid": "",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "4",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": "",
        "_mcu": {
            "mcu": false
        }
    }
]
```

phoddie, 2022-10-26:
Thank you! That works with the changes made to support groups. My hope is to get that posted late today, but perhaps tomorrow.

FWIW – the MQTT In node above uses `auto-detect` for the data format. That's not yet implemented, so I changed it to `utf8`. That let me test with the broker at `test.mosquitto.org` (which sends lots of "test" data!).

phoddie, 2022-10-27:
Fix committed.

ralphwetzel, 2022-11-01:
👍 Confirmed.

ralphwetzel, 2022-11-01:
👍 Confirmed!

closed 2022-11-01 by ralphwetzel