Moddable-OpenSource/moddable #955: Timer.schedule with no repeat or interval delete the timer on Windows

brenocastrocardoso, 2022-11-03:
**Build environment:** Windows
**Target device:** desktop simulator

**Description**
The Timer.schedule function when called without repeat or interval will make the timer NULL on windows while in MAC and Esp32 plataforms it will unschedule the timer. As described in the documentation:

>  If called with neither interval nor repeat arguments, the timer is unscheduled and will not trigger until rescheduled using Timer.schedule.

So this code:

```javascript

let debounceTimer;

export default function () {

  debounceTimer = Timer.set(() => {
    trace(`Debounce Timer run\n`);
    // After the timer run, unschedule it
    Timer.schedule(debounceTimer);
  },1000);

  // Timer used to simulate an user event
  Timer.set(() => {
    // When "user event" arrive schedule the timer
    Timer.schedule(debounceTimer, 1000);
  }, 5000, 5000);
}
``` 

Will break on Windows with the output:
```
Debounce Timer run
<<Filepath>> # Break: (host): C: xsGetHostData: 
```
And will work just as expected on MAC with the output being:
```
Debounce Timer run
Debounce Timer run
Debounce Timer run
Debounce Timer run
```

**Steps to Reproduce**
1. Run the code above on windows

**Additional Information**
I was assuming the issue was calling Timer.schedule without repeat and interval and after that try to schedule the same timer. But the code bellow don't break. 

```javascript
  let debounceTimer = Timer.set(() => {
    trace(`Debounce Timer run\n`)
  },1000);
  Timer.schedule(debounceTimer);
  Timer.schedule(debounceTimer, 1000, 1000);
```

andycarle, 2022-11-03:
Thank you for the report. I can confirm the issue on Windows and will look into solutions today.

andycarle, 2022-11-03:
I've committed a fix for this issue locally. It will be included in our next GitHub release.

andycarle, 2022-11-04:
This fix is now live. @brenocastrocardoso, please give it a try when you can and let us know if the issue is resolved in your application.

brenocastrocardoso, 2022-11-04:
Hi Andy, thanks very much for the fast fix! I don't have access to an windows machine at the moment I will have that tested on Monday and let you know

brenocastrocardoso, 2022-11-07:
Hi Andy, I tested it here and I'm still seeing some issue, using the same code as before:

```javascript 
let debounceTimer;

export default function () {

  debounceTimer = Timer.set(() => {
    trace(`Debounce Timer run\n`);
    // After the timer run, unschedule it
    Timer.schedule(debounceTimer);
  },1000);

  // Timer used to simulate an user event
  Timer.set(() => {
    // When "user event" arrive schedule the timer
    Timer.schedule(debounceTimer, 1000);
  }, 5000, 5000);
}
```

Using  the windows simulator it seems that the timer is not being rescheduled after being unscheduled. The windows simulator will output just one time "Debounce Timer run". While Mac simulator will output it every 5 seconds which is the expected behavior.

andycarle, 2022-11-07:
@brenocastrocardoso You are correct; there is an additional issue that is surfaced by this example. I'll get another fix out this week.

andycarle, 2022-11-10:
@brenocastrocardoso I have pushed another set of changes around Windows timers. 

edf0c3d directly addresses the problem you raised but is actually now outdated. In b80a3d3f420785055ae28e3578a8838d059a46a5 I decided to completely rework the Windows Timer implementation to use a much more accurate timer API. The Windows Timer module is now passing all timer tests in [testmc](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/testing.md) except for one, which fails because of a hard cap of 16 concurrent Timers on Windows.

I've also added a new test case based on this issue (see ed46949).

andycarle, 2022-11-22:
Hi @brenocastrocardoso,

Have you had a chance to try this out? I believe everything here has been resolved.

closed 2022-11-29 by phoddie