Moddable-OpenSource/moddable #956: CRC16 inconsistency?

ralphwetzel, 2022-11-05:
According to (my interpretation of) the documentation of [CRC16.checksum()](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/data/data.md#checksumbuffer) it shouldn't matter if a buffer (to be processed) is sent at one shot or in several packages; as long as data is identical in values & sequence, the result should be the same.

This yet isn't the case in this example:

``` javascript
function test_crc() {
    let a1 = Uint8Array.of(0xf0, 0x88, 0);
    let a2 = Uint8Array.of(255,254,0,0,0,136,255,255);
    let a3 = new Uint8Array([...a1, ...a2]);
    let c;
    
    let crc16 = new CRC16(0x8005, 0, true, true, 0xFFFF);

    crc16.reset();
    crc16.checksum(a1.buffer);
    c = crc16.checksum(a2.buffer);
    trace("a1, then a2: " + c.toString(16) + "\n"); //a664 => incorrect!

    crc16.reset();
    c = crc16.checksum(a3.buffer);
    trace("a3: " + c.toString(16) + "\n");  // afab => correct!

}
```

What am I doing wrong?

phoddie, 2022-11-06:
Looks like a real bug. Thanks for the report.

Just FYI – since the CRC module uses the [latest native buffer APIs](https://blog.moddable.com/blog/buffers/) you can pas `a1` instead of `a1.buffer` to the `checksum` method.

phoddie, 2022-11-09:
This should be better now. We also added a test case to help it stay fixed. ;)

closed 2022-11-29 by phoddie