Moddable-OpenSource/moddable #964: NimBLE writeNoResponse not working on esp32

FWeinb, 2022-11-21:
**Build environment:** macOS
**Target device:** esp32/m5atom-matrix

**Description**
When building a BLE Project with the moddable sdk including "writeNoResponse" in "properties" if an GATT Service declaration json file, will alter the generated `.flags` declaration in the `mc.bleservices.c` but has no effect when run on the device

**Steps to Reproduce**
1. Build this [demo gist](https://gist.github.com/FWeinb/e4f4400301494ad5613fbf2b80e631ab) and deploy on esp32
2. Go to https://googlechrome.github.io/samples/web-bluetooth/characteristic-properties.html?service=04b5a505-6be3-4408-9eb5-e93dc0b32255&characteristic=ef3cb3d3-7097-4f99-b3c5-b5018e3d80b0

Compare with this output
```
Requesting Bluetooth Device...
Connecting to GATT Server...
Getting Service...
Getting Characteristic...
> Characteristic UUID:  ef3cb3d3-7097-4f99-b3c5-b5018e3d80b0
> Broadcast:            false
> Read:                 true
> Write w/o response:   false <--- Should be `true`
> Write:                true
> Notify:               false
> Indicate:             false
> Signed Write:         false
> Queued Write:         false
> Writable Auxiliaries: false
```

**Expected behavior**
It should be possible to enable the `Write w/o response` property. 

**Other information**
I can confirm that using the [NimBLE Arduino](https://github.com/h2zero/NimBLE-Arduino) library and setting up a Characteristc via the C++ API is working on the hardware.

andycarle, 2022-11-21:
Thank you for the report (and for the nifty way to test it). I'll take a look at our implementation here later today.

andycarle, 2022-11-22:
@FWeinb Unfortunately, I'm not able to reproduce this issue. When I set the device up as instructed and run the test at https://googlechrome.github.io/samples/web-bluetooth/characteristic-properties.html?service=04b5a505-6be3-4408-9eb5-e93dc0b32255&characteristic=ef3cb3d3-7097-4f99-b3c5-b5018e3d80b0 this is what I get:

[image]

So, now we need to figure out what's different between my build and yours.

One easy possibility: is there any chance that you have stale build output that wasn't updated? I've seen occasional issues where `mc.bleservices.c` will be updated but that update is not propagated through to trigger a rebuild of `modBLEServer.c`. You could rule this out by completely deleting `$MODDABLE/build/bin/esp32` and `$MODDABLE/build/tmp/esp32` and trying your build/deploy again. 

If that fixes the problem, I could use some more information about your build environment so I can hunt down why you're getting a stale build deployed. If it does not, we have a different kind of hunting to do...

phoddie, 2022-11-22:
@andycarle – haven't we also seen situations where a computer / phone will cache information about a BLE device? I remember it being remarkably difficult to change the BLE device name because of that.

andycarle, 2022-11-22:
We have seen caching issues, yes. I did not see those problems in Edge on Windows today, but it is definitely a possibility.

FWeinb, 2022-11-22:
@andycarle @phoddie Thanks for the fast investigation, it was a caching issue! It is not cached in the browser level but  on an OS level (macOS 13.0.1 (22A400)). I had to turn bluetooth off/on to get these values updated.