Moddable-OpenSource/moddable #967: Example wifi-config crashes with "invalid color"

rei-vilo, 2022-11-27:
**Build environment:** macOS
**Target device:** Moddable Two, both board and simulation

**Description**

Example wifi-config crashes with "invalid color".

> /Users/USER/Projects/moddable/modules/piu/All/piuColor.c (147) # Break: Skin: invalid color!

**Steps to Reproduce**

1. Command line

```
cd /Users/USER/Projects/moddable/examples/piu/wifi-config
% mcconfig -d -m -p sim/moddable_two
% mcconfig -d -m -p esp32/moddable_two
```

3. The screen displays a list of SSIDs. 
4. Selecting one crashes with the message displayed above.

**Expected behavior**
No error message

**Images**

[image]

**Other information**

Thank you!

phoddie, 2022-11-29:
Thanks for the report. Somehow this got broken along the way. It will be fixed in the December Moddable SDK release. The change is to add a `color` property to the style in line 28 of `$MODDABLE/examples/piu/wifi-config/assets.js`:

```js
const OpenSans20 = Style.template({ font: "20px Open Sans", color: BLACK, vertical: "middle", horizontal: "left" });
```

rei-vilo, 2022-11-29:
Strange, there is already a `OpenSans20` constant defined. 

https://github.com/Moddable-OpenSource/moddable/blob/758b624dc985362c77cd283739bec7f87bc28ac2/examples/piu/wifi-config/assets.js#L26

The edited code...

```
const OpenSans20 = Style.template({ font: "20px Open Sans", vertical: "middle", horizontal: "left" });
const OpenSans20 = Style.template({ font: "20px Open Sans", color: BLACK, vertical: "middle", horizontal: "left" });
```

results in an error.

```
/Users/USER/Projects/moddable/examples/piu/wifi-config/assets.js:29: error: duplicate variable OpenSans20!
```

rei-vilo, 2022-11-29:
Great, it works fine now.

closed 2022-11-29 by rei-vilo