Moddable-OpenSource/moddable #968: Archive SDKs for faster build

rei-vilo, 2022-11-27:
**Describe**

`mcconfig` builds the whole SDKs for each example.

Couldn't the SDKs be archived (`.a`) and shared among the different projects targeting the same board?

The SDKs include the Moddable SDK and ESP32 SDK (1033 files)

```
[461/1033] Building C object esp-idf/heap/CMa...__idf_heap.dir/port/memory_layout_ut
``

**Why do you think this feature would be useful?**

Save time and space.

**Describe alternatives you've considered**

None so far

andycarle, 2022-11-28:
Thank you for the suggestion! 

This is possible to some extent, yes. The Moddable SDK ESP32 build formerly did something similar by using a shared ESP-IDF build directory between all projects.

The challenge is that various Moddable SDK projects for ESP32 use differing SDKCONFIG options and partition tables. Some also use custom defines that ripple through to the ESP-IDF build. Those configuration changes make it necessary to reconfigure and rebuild the ESP-IDF when switching between some projects. Those reconfiguration operations were a common source of difficult-to-isolate bugs and developer confusion, so we eventually went to the current system of independent builds for each project.

The workflow that suffers most dramatically from this decision is building/deploying multiple examples in a row. I agree that it would be nice to find a better solution there (starting out in the desktop simulator is one somewhat unsatisfying answer). We find this issue to be much less problematic once a developer starts diving into their own project and primarily rebuilding just that one project.

(I just noticed that you also mentioned the Moddable SDK build here. It has a similar challenge: configuration differences between project manifests impact the build of the Moddable SDK.)

I do get where you're coming from here, though. And I think we'd welcome suggestions or improvements to the build process to speed things up when moving between example projects!

tve, 2022-12-15:
I just happened to bump into this not long after posting #987 "A different approach to hello world!". That would be a solution... I.e., a base image that has most of the kitchen sink thrown in and most of the examples as mods. Users would not need esp-idf and would not need to build a full image. Just compile to bytecode and install the mod (IIUC)...

phoddie, 2022-12-15:
A "kitchen sink" may work for many ESP32 devices, but there are many MCUs (ESP8266 is a good example) which don't have sufficient flash to hold everything.