Moddable-OpenSource/moddable #975: Error using esp32/esp32c3

louisvangeldrop, 2022-12-12:
# Coredump when loading Helloworld example with esp32c3 board

**Build environment:**  Windows
**Target device:** ESP32C3

**Description**
After a successful upload of the example, the board shows a coredump:
```
Coredump checksum='5cbca8c3'
Rebooting...
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.
Core  0 register dump:
MEPC    : 0x42019204  RA      : 0x4201905e  SP      : 0x3fc93c80  GP      : 0x3fc8bc00
TP      : 0x3fc7da10  T0      : 0x4005890e  T1      : 0x420244e2  T2      : 0x0000000a
S0/FP   : 0x3fc9c038  S1      : 0x3c05c6d8  A0      : 0x3fc9c20c  A1      : 0x00000000
A2      : 0x00000001  A3      : 0x00000001  A4      : 0x3c053000  A5      : 0x42019904
A6      : 0x74656772  A7      : 0x3fc93dfd  S2      : 0x00000030  S3      : 0x3fc8d000
S4      : 0x3fc9c0c8  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x0a0d0000  T4      : 0x00000053  T5      : 0x0000002a  T6      : 0x00000020
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000002  MTVAL   : 0x5c873787
MHARTID : 0x00000000
Stack memory:
3fc93c80: 0x3fc9c20c 0x3c056cfc 0x3fc9c038 0x3fc9c068 0x3c060d08 0x3fc9c038 0x0000000c 0x00000001
3fc93ca0: 0x3fc9c038 0x00000000 0x00000000 0x3fc93ef0 0x00000000 0x00000076 0x00000001 0x00000000
3fc93cc0: 0x0a0d0000 0x00000053 0x3fc9164c 0x403855be 0x00000000 0x00000000 0x3fc9164c 0x40385be0
3fc93ce0: 0x00000009 0x0000003b 0x00000063 0x00000000 0x00000000 0x00000000 0x00000028 0x40387556
3fc93d00: 0x00000000 0x00000000 0x000000d8 0x00000000 0x00000000 0x00000000 0x3fc9c888 0x403855be
3fc93d20: 0x00000000 0x00000000 0x3fc9c888 0x40385be0 0x00000000 0x00000000 0x00000000 0x00000000
3fc93d40: 0x00000000 0x00000000 0x00000000 0x3c052fac 0x00000000 0x00000000 0x00000000 0x3c052fac
3fc93d60: 0x42007872 0x420060ac 0x3fc93de0 0x3fc8bc00 0x3fc7da10 0x4005890e 0x00000000 0x0000000a
3fc93d80: 0x3fc9c20c 0x3c053000 0x3fc9c20c 0xffffffff 0x3c0530b8 0x00000242 0x00000006 0x00000000
3fc93da0: 0x3fc93ef0 0x3fc9c0a8 0x3fc9c0c8 0x3fc9c0e8 0x00000000 0x00000000 0x00000000 0x00000000
3fc93dc0: 0x00000000 0x00000000 0x3fc9c0c8 0x3fc8d000 0x3c053000 0x3c053000 0x3fc9c20c 0x420061c0
3fc93de0: 0x00000000 0x3fc9c20c 0x3c061578 0x00000001 0x75746573 0x61742f70 0x74656772 0x00000000
3fc93e00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93e20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93e40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93e60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93e80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93ea0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93ec0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93ee0: 0x00000000 0x00000000 0x00000000 0x00000000 0x42005fc2 0x3fc8d000 0x00000000 0x00000000
3fc93f00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93f20: 0x00000000 0x3fc93de0 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93f40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93f60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93f80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93fa0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93fc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93fe0: 0x00000000 0x0000025c 0x00000000 0x00000000 0x3c0613d8 0x3c052acc 0x00008000 0x42005f12
3fc94000: 0x00000000 0x00000000 0x3c061418 0x3fc94208 0x3fc9c208 0x00000600 0x00000200 0x00000200
3fc94020: 0x00000000 0x3fc9c138 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00008000
3fc94040: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc8d000 0x42004f54
3fc94060: 0x00000000 0x00000000 0x00000000 0x40387c4c 0x00000000 0x00000000 0x00000000 0x00000000
ELF file SHA256: b1ac6dfa869f88db
=
```

After the coredump the board reboots and  so on

mkellner, 2022-12-16:
I'm sorry for your inconvenience. Please use esp-idf-4.4.2 until we resolve the issue with version 4.4.3. Thank you.

louisvangeldrop, 2022-12-17:
Ok. I can wait until the issue is resolved.

closed 2023-01-11 by mkellner

phoddie, 2023-01-11:
@louisvangeldrop - if you have a chance to confirm that this fixes the problem for you, that would be very helpful. Thank you!

louisvangeldrop, 2023-01-12:
It is working now like a charm. Thanks a lot for your beautiful work.

phoddie, 2023-01-12:
Excellent! Thank you for the quick confirmation.