Moddable-OpenSource/moddable #979: Linux installation fails: bash scripts have literal \n instead of newlines

tve, 2022-12-13:
**Build environment:** Linux arch

**Description**
The shell scripts built have literal \n's in them

**Steps to Reproduce**
1. Git clone repo
2. cd build/makefiles/lin
3. make
```
make -f /home/sw/moddable/moddable/xs/makefiles/lin/xsc.mk                           
make[1]: Entering directory '/home/sw/moddable/moddable/build/makefiles/lin'         
# xsc debug : cc xsBigInt.c                                                          
# xsc debug : cc xsCode.c                                                            
...
# tools debug : mcconfig
...
# tools debug : mcrun
make[1]: Leaving directory '/home/sw/moddable/moddable/build/makefiles/lin'
/home/sw/moddable/moddable/build/bin/lin/debug/mcconfig -d -m -p x-lin /home/sw/moddable/moddable/tools/xsbug/manifest.json
make: /home/sw/moddable/moddable/build/bin/lin/debug/mcconfig: No such file or directory
make: *** [makefile:42: debug] Error 127
```
4. cat /home/sw/moddable/moddable/build/bin/lin/debug/mcconfig
```
#!/bin/bash\nDIR=$(cd "$(dirname "$BASH_SOURCE")"; cd -P "$(dirname "$(readlink "$BASH_SOURCE" || echo .)")"; pwd)\n$DIR/tools mcconfig "$@"
```
5. Notice that this is one line with embedded literal \n's, that ain't gonna work...

andycarle, 2022-12-13:
Interesting... This is what I get in zsh on Ubuntu:

```console
 acarle@ANDY-SURFACE ~/Projects/moddable cat ./build/bin/lin/debug/mcconfig
#!/bin/bash
DIR=$(cd "$(dirname "$BASH_SOURCE")"; cd -P "$(dirname "$(readlink "$BASH_SOURCE" || echo .)")"; pwd)
$DIR/tools mcconfig "$@"
```

andycarle, 2022-12-13:
A few thoughts.

1. As you probably noted, these scripts are generated in a Makefile (`build\makefiles\lin\tools.mk`).
2. I believe `make` uses `/bin/sh` by default. 
3. `sh` does not have a `-e` option and always interprets backslash escapes.

Perhaps you either have `/bin/sh` mapped so something else (e.g. `fish`) or you have over ridden the default shell for `make`?

tve, 2022-12-13:
It works on ubuntu... (Although I have a headless system and didn't appreciate having 177 packages installed for libgtk-3-dev)
On Arch /bin/sh seems to have been bash for a long time
On Ubuntu /bin/sh is dash
I would suggest that adding `-e` to the `echo` statements is more portable... Don't you need that on macos as well? I have some vague (painful) memories of deja-vu...
I installed dash and ran `make SHELL=/usr/bin/dash` and that seems to have completed the build.
Thanks for the prompt replies!

andycarle, 2022-12-13:
You're right, `dash` is what is being invoked on my system. 

This from `dash`'s `echo` seems problematic to the approach of always adding the `-e`:

```console
$ echo -e "Testing\n123"
-e Testing
123
```

tve, 2022-12-13:
then maybe set SHELL=dash in the makefile?

It looks like what I built compiles the helloworld fine, but I'm not sure it ran fine. When I ran `mcconfig -d -m -p lin` xsbug immediately displayed a stack trace on `readPreferenceString` and in its console shows:
/home/sw/moddable/moddable/modules/piu/PC/lin/piuFiles.c (222) # Break: readPreferenceString: No such file or directory!
Dunno whether I skipped some instructions or whether instructions are incomplete...

andycarle, 2022-12-13:
May be another difference with arch — we definitely do most of our Linux testing on Ubuntu. I recommend opening a new issue for that one and I'll get the right person pointed to it.

andycarle, 2022-12-29:
I think setting `SHELL = /bin/dash` in `tools.mk` is a reasonable compromise here. On the odd system that doesn't have `/bin/dash` that should at least fail fast. 

I'll get that change in.

closed 2023-01-10 by phoddie

tve, 2023-01-23:
Works now. Thanks!