Moddable-OpenSource/moddable #981: Helloworld example fails on freshly built moddable SDK

tve, 2022-12-13:
**Build environment:** Linux **Arch** x64
**Target device:** desktop simulator,

**Description**
A brief summary of the problem

**Steps to Reproduce**
1. Build and install moddable SDK for linux following the instructions, except that on Arch I have to use `make SHELL=/usr/bin/dash`
2. Start `xsbug`
3. Run `mcconfig -d -m -p lin`
4. Observe that xsbug immediately displays a stack trace on `readPreferenceString` and in its console shows:
```
/home/sw/moddable/moddable/modules/piu/PC/lin/piuFiles.c (222) # Break: readPreferenceString: No such file or directory!
```

I'm totally new to the moddable SDK, so most lilely it's me doing something wrong...
[image]

mkellner, 2022-12-13:
Can you try running `mcconfig -d -m -p lin` a second time?

tve, 2022-12-13:
same result (I'm on Gitter if that's convenient)

tve, 2022-12-13:
So the answer is that I has started xsbug before starting the helloworld example. Stopping xsbug, re-running mcconfig brought up xsbug that without the error.
The getting started instructions kind'a lead to this situation: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md?plain=1#L367-L378

mkellner, 2022-12-13:
I am looking further into the cause of the problem. There should be no problem starting with or without xsbug running (as the build will start xsbug if necessary). I will update this issue when I have more information.

I'm glad you have a workaround for now.

phoddie, 2022-12-16:
This exception has been [quieted](https://github.com/Moddable-OpenSource/moddable/commit/a5d03c918b17c644710322f1a2326e8f0f24ea3d).

closed 2022-12-27 by phoddie