Moddable-OpenSource/moddable #985: [Driver Support] BME680

Sineos, 2022-12-14:
**Describe**
Support for the BME680 sensors.
Just starting to dig into this concept and it sounds really promising. 
Since I have some BME680 lying around, I was wondering if I could setup some kind of room thermostat with a small 2.4" display, an ESP32 and a BME680 and have it running in node-red.

andycarle, 2022-12-14:
Hi @Sineos,

Yes, this should certainly be possible. We have support for a number of ambient environment and gas sensors already, but not that particular sensor. 

I'll pick one up and see what I can do with it. Do you have any experience with driver development that you would like to put towards getting this working?

Sineos, 2022-12-15:
Thanks @andycarle for considering.
As much as I would love to contribute, my development skills are unfortunately quite weak and my understanding of hardware registers etc. comparable to my knowledge on quantum mechanics 🤷‍♂️

Sineos, 2022-12-15:
I had a deep look at https://github.com/adafruit/Adafruit_CircuitPython_BME680/blob/main/adafruit_bme680.py and https://github.com/Moddable-OpenSource/moddable/blob/public/modules/drivers/sensors/bmp280/bmp280.js

Then back to quantum mechanics:
[image]

phoddie, 2023-05-31:
Closing this out as the [BM68x driver](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/drivers/sensors/bme68x/bme68x.js) is now in available along with [some example](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/drivers/sensors/bme68x).

closed 2023-05-31 by phoddie